DeviationTX/deviation #219: Unstaged changes

vladislavy, 2017-03-19:
Now I have two unstaged changes even if I don't change anything:

Binary files a/install_nightlies.tar.enc and b/install_nightlies.tar.enc differ
src/target/common/devo/msc2/lib/stm32f10x_flash.c

I think it can be related to:
https://github.com/DeviationTX/deviation/commit/cfb19f6fba2423589bf1834037232eb74ce06641

goebish, 2017-03-19:
I'm also having an issue with src/target/common/devo/msc2/lib/stm32f10x_flash.c

TheRealMoeder, 2017-03-19:
When I pulled in upstream for Extended-Audio builds GIT Auto corrected stm32f10x_flash.c for me (it has something to do with the line ends), but i cannot remember the exact message git merge gave me. Anyways, my next push updated the file, which can be found here. https://github.com/TheRealMoeder/deviation/blob/4ecf8a22bbb58c8b219b96eb9fc8d15fe023ebb6/src/target/common/devo/msc2/lib/stm32f10x_flash.c

F-D-R, 2017-03-19:
Yes, that is because of the line ends.
Probably those files had CrLf line ends somewhere, and now git checks out everything with Lf only line ends, so it seems like it was changed.
If I remember well, I had to checkout those files...

F-D-R, 2017-03-19:
Well, you need to reset the existing files to reflect the new line endings:

git rm -rf --cached .
git reset --hard HEAD

goebish, 2017-03-19:
This is even worse now, I might better delete my repo and fork again ... but I'll lose my history and branches.

TheRealMoeder, 2017-03-19:
Why is it worse? What errors do you get?

goebish, 2017-03-19:
I can't even switch branches now
error: Your local changes to the following files would be overwritten by checkout:
	src/target/common/devo/msc2/lib/stm32f10x_flash.c
Please, commit your changes or stash them before you can switch branches.
Aborting

TheRealMoeder, 2017-03-19:
What has git diff to offer on this file?

TheRealMoeder, 2017-03-19:
In case of doubt, either 
git add src/target/common/devo/msc2/lib/stm32f10x_flash.c
If only line endings changed this won't hurt

goebish, 2017-03-19:
That kinda works, but I've to do it every time I want to switch from this branch.
This is not the first time I'm having issues with line endings and git.
Oh well, I think I'll end up forking again, I already lost a couple of hours trying to fix the issue, I'm losing my nerves ;)

goebish, 2017-03-19:
$ git diff
diff --git a/src/target/common/devo/msc2/lib/stm32f10x_flash.c b/src/target/common/devo/msc2/lib/stm32f10x_flaindex 31ca625..cdff9e9 100755
--- a/src/target/common/devo/msc2/lib/stm32f10x_flash.c
+++ b/src/target/common/devo/msc2/lib/stm32f10x_flash.c
@@ -710,7 +710,7 @@ FLASH_Status FLASH_ProgramWord(uint32_t Address, uint32_t Data)
         *(__IO uint16_t*) tmp = Data >> 16;

         /* Wait for last operation to be completed */
-        status = FLASH_WaitForLastOperation(ProgramTimeout);
+        status = FLASH_WaitForLastOperation(ProgramTimeout);

         /* Disable the PG Bit */
         FLASH->CR &= CR_PG_Reset;
warning: CRLF will be replaced by LF in src/target/common/devo/msc2/lib/stm32f10x_flash.c.
The file will have its original line endings in your working directory.

TheRealMoeder, 2017-03-19:
Well obviously the CRLF is wrong in the repository, so it makes sensing doing the add in your fork as well (if my voice alert pull request gets merged it will be fixed in upstream/master as well). Concerning the branches you work on, you should merge in your forks master afterwards to reflect the new file. But I agree this is annoying, but it is rather windows' than git's fault ;)

goebish, 2017-03-21:
I think my last PR should fix it as well
https://github.com/DeviationTX/deviation/pull/223

goebish, 2017-03-21:
I confirm it's fixed on my side.
You might have to delete stm32f10x_flash.c in your local repo before chekout.

closed 2017-03-22 by TheRealMoeder