DeviationTX/deviation #282: Load model and change icon, DEVO10 will restart

huaitao, 2017-07-26:
Load model and change icon, DEVO10 will restart.(The modelico folder contains 87 icons)
Edition:debug-devo10-v5.0.0-d9cd81f

vladislavy, 2017-07-31:
Will it work normally if you decrease number of icons? What time it takes to open the list of icons? From pressing the button to displaying the list.

huaitao, 2017-08-02:
When decrease number of icons it work normally.If do not decrease number of icons,it takes about 2 seconds.(Some mode can change icon and not restart)

vladislavy, 2017-08-02:
OK, it looks like I know the reason.

vladislavy, 2017-08-02:
Can you test the build with fix?
https://www.deviationtx.com/forum/builds/7261-load-model-and-change-icon-devo10-will-restart

huaitao, 2017-08-02:
OK,I'll test tonight.

huaitao, 2017-08-03:
Sorry,I tried, but it didn't work，but I downloaded the latest version and found no this problem(deviation-devo10-v5.0.0-9c46f3b).I think it was caused by the wrong modelx.ini configuration file.Sorry,I should have tested and asked questions.Thank you for what you've done.

closed 2017-08-03 by vladislavy