DeviationTX/deviation #285: branch5 make TARGET=devof7 error

bartremes, 2017-08-04:
make TARGET=devof7
results in this error 

ERROR: Target 'en_US' is not in list (devo8 devo10 devo12)
make: *** [filesystem/devof7/models/model10.ini] Error 1

bartremes, 2017-08-04:
#define NO_LANGUAGE_SUPPORT 1
in target_def.h 
is the problem
also devof4 and devof12e have the same problem

so I think in language.c 
#ifdef NO_LANGUAGE_SUPPORT 
is not working correct

TheRealMoeder, 2017-08-04:
Travis seems to build it just fine, for example f12e:
https://travis-ci.org/DeviationTX/deviation/jobs/259667721

Did you pull master or anything else? What is your build environment?

bartremes, 2017-08-04:
I also build a fresh master.  But without success.

TheRealMoeder, 2017-08-04:
Instead of
`make TARGET=devof7e`
please run 
`make devof7e`

bartremes, 2017-08-04:
thanks 
`make devof7 `
and
`make devof7 zip` 
works
I will now test the build on my devof7 with extra memory

bartremes, 2017-08-04:
hmm
the protocol map is not in the zip file.
all the rest is there except for the protocol map.

During the zip build they are build:
 + Compiling 'protocol/assan_nrf24l01.c'
 + Compiling 'protocol/bayang_nrf24l01.c'
 + Compiling 'protocol/bluefly_nrf24l01.c'
 + Compiling 'protocol/cflie_nrf24l01.c'
....

Any help is welcome

PhracturedBlue, 2017-08-04:
the f7 doesn't have enough memory to include the protocols, so they are built as modules and must be manually chosen and installed.  If you have a larger processor, you should be able to comment out the MODULAR line in devof7/Makefile.inc to build all modules into the binary (see the difference between 'devo7e-256' and 'devo7e').  I haven't tested this, and it has been a while since I've looked at it, so I might have missed something.

bartremes, 2017-08-04:
Solved!
I had a look to the Makefile.inc of 'devo7e-256' and 'devo7e' and had to add to mine Makefile.inc of the devof7 
LANGUAGE    := devo10

now I can compile again whit 
make TARGET=devof7
make TARGET=devof7 zip 

other things I did because of my bigger memory are:

in: 
deviation/src/target/devof7/target_defs.h 
change the size of the memory to 512 
#define USE_DEVOFS 0 //Must be added before #include "../common/devo/common_devo.h" 
and 
#define SPIFLASH_SECTORS 512 
//#define SPIFLASH_AAI_AF 1 
save 

change in: 
deviation/src/target/devof7/Makefile.inc 
add under
FILESYSTEMS := common text
add:
LANGUAGE    := devo10
to make shure you dont make use of the compressed upload method. Now you will have the original filesystem 
in the zip 
#PRE_FS = $(ODIR)/.pre_fs 
#INCLUDE_FS := 0 
include target/common/devo/Makefile.inc 
#include target/common/devo/Makefile.devofs.inc 

Thanks for all the help!!

closed 2017-08-05 by bartremes