DeviationTX/deviation #311: FrskyX Timing

Wene001, 2017-10-31:
i get very sporadic telemetry with FrskyX in LBT EU mode.
First i had the same problems with multimodule, but (thx to) MikeBlandford got it fixed.
https://www.rcgroups.com/forums/showpost.php?p=38484818&postcount=10107

Could these be tested and implemented in Deviation, please
Cheers Wene

goebish, 2017-11-12:
Please check this build and report:
https://www.dropbox.com/sh/h1ait0i6kwgf6q3/AACKUCvch2TdpJRHGC1MAb3Ca?dl=0

Wene001, 2017-11-12:
Thanks, but i cant test 7e or 10 Builds
My Hardware is Devo6s, Devo12s and DevoF7_XMS

I'll test this 
https://www.deviationtx.com/forum/protocol-development/5356-new-frskyx-protocol?start=880#65760
if i had free time

goebish, 2017-11-12:
Ah ok, I had not noticed hexfet has already fixed it.

Wene001, 2017-11-12:
Made a short workbench test with a X6R in LBT Mode.
Link quality an telemetry stability looks fine now.

TheRealMoeder, 2017-11-18:
Fixed in #323

closed 2017-11-18 by TheRealMoeder