DeviationTX/deviation #314: Can't Bind DEVO RX1002

Chen-Leon, 2017-11-09:
Nightly build version can't Bind DEVO RX1002

Chen-Leon, 2017-11-13:
YOUTUBE VIDEO

V5.0.0
https://www.youtube.com/watch?v=GoNoLwbtsKA

Nightly Builds 171111
V5.0.0-6da40db
https://www.youtube.com/watch?v=m24pvipYmDs

hexfet, 2017-12-09:
According to the videos it's 5.0.0 that's not binding.  Just tried with my Devo 10 and bound with both 5.0.0 and the latest nightly build.  Not able to duplicate the problem.

JacekHoleczek, 2017-12-10:
I tried my DEVO 10 with the latest nightly build "v5.0.0-711ada2" (dated 2017-12-08) and I was able to bind my RX1002 receiver without problems (I tried several times).

Two things you can check.

1. In the "Model setup" submenu (just below the "Fixed ID" field) there is the "# Channels" field ... make sure that its value is "10" (it will not bind if the number of channels is wrong).

2. In the beginning of your video, when the "Fixed ID" is set to "123456", the button next to the "DEVO" spin-box should say "Bind" but then, after you clear the "Fixed ID" (set to "None"), it should change to "Re-Init". I do not speak your language but it seems to me that, in your video, in both cases it displays the same character string (is the "Fixed ID" really set to "None"?).

hexfet, 2017-12-10:
I'm able to bind with the number of channels set to other values.  What I did notice is the number of channels cannot be changed after binding.  After changing the number of channels and reloading the model I had to re-bind (or change the number of channels back to what it was bound with and reload).

JacekHoleczek, 2017-12-11:
Many thanks for the correction.
I repeated my trials and I can confirm your findings.
I am also able to bind the receiver with an "incorrect" number of channels. However, once the receiver is bound with some "Fixed ID" set, changing the number of channels requires a "Re-Init".

hexfet, 2017-12-11:
Closing - Unable to duplicate.  Two reports of binding successfully with nightly build.

closed 2017-12-11 by hexfet

Chen-Leon, 2017-12-12:
Thank's hexfet JacekHoleczek F-D-R
I am so sorry , It's my mistake