DeviationTX/deviation #330: Main page text

Chen-Leon, 2017-12-13:
Build version is 711ada2
download the dfu on deviation web site
Tx Type is Devo10
Can tried reproducing this on the emulator

[image]

vladislavy, 2017-12-13:
I think it related to "Modifying 04b03 font to have hungarian characters"
https://github.com/DeviationTX/deviation/commit/49dfbca2711a6f8d3754abe2257a0cd9c3df18c
Before it all looks fine.

F-D-R, 2017-12-13:
Strange, I didn't change anything about alignment, just added a few characters, and it worked well in the 12E emulator and I think I tried it in the TX  as well, but I have to check that...

vladislavy, 2017-12-13:
I just change 04b03 font at emulator to old version and the alignment changed. It'a a miracle! I don't understand how it possible.

F-D-R, 2017-12-13:
I guess FontForge does some changes in the background.
I'm quite unhappy with it anyway, because I could ruin the font by simple edit commands, like copy/paste/cut/undo etc, so in the end I had to be very careful with changes to get a working font...

F-D-R, 2017-12-13:
BTW how you can use this font for Russian? There are no Russian characters in it, but it is used for translated text on the trim page...

vladislavy, 2017-12-14:
I don't use Russian with transmitters at all.

vladislavy, 2017-12-14:
I think it related to some problem with alignment at "bdf_to_font.pl"

Before we havn't any symbol with 6 pixel height without descent in charset.

vladislavy, 2017-12-14:
Just now tested it. I manually added 1 pixel at top to exclamation mark (BBX 1 6 0 0) with old 04b03.bdf and get the same result.

F-D-R, 2017-12-14:
I see.
What to do? Those characters need those pixels. I even had to lower all the capital letters to make some room...

vladislavy, 2017-12-15:
Nothing. Apply the fix and forget about it. I think it will never change in the future, now all pixel places (8 pixel height) are used. But takes in mind that it can happen with new additional fonts.

closed 2017-12-15 by vladislavy