DeviationTX/deviation #354: Is it maybe a BUG is the software of "Deviation"? (FrSky)

Antennenfreund, 2018-02-02:
Hello dear friends!

With Frsky receivers there are unfortunately very strong range issues. Me and a friend of mine get a failsave after like 50-100 meters with Frsky!

"FrSky R-XSR Ultra SBUS / CPPM D16 16CH Mini Redundancy Receiver 1.5g for RC Multirotor FPV Racing Drone"

https://www.banggood.com/FrSky-R-XSR-Ultra-SBUS-CPPM-D16-16CH-Mini-Redundancy-Receiver-1-5g-for-RC-Multirotor-FPV-Racing-Drone-p-1186057.html?rmmds=search

Does anyone else have the problem?

On "afdhs 2a" there were no range issues so far with an x6b reciever Unfortunately the receiver is too big for some builds.

This is the one: "Flysky X6B 2.4G 6CH i-BUS PPM PWM Receiver for AFHDS i10 i6s i6 i6x i4x Transmitter"

https://www.banggood.com/Flysky-X6B-2_4G-6CH-i-BUS-PPM-PWM-Receiver-for-AFHDS-i6s-i6-i6x-Transmitter-p-1101513.html?rmmds=search&cur_warehouse= USA

Are there any alternative for small recievers with rssi telemetry?

Is it maybe a BUG is the software of "Deviation"? (FrSky)

There is also a problem with the included reciever, it only works in ppm, not sbus or ibus mode. This adds the issue that you cant set up a failsave properly and the quad stays armed, motors spinning so using that is just dangerous.

https://www.banggood.com/818CH-Mini-Receiver-With-PPM-iBus-SBUS-Output-for-Flysky-i6-i6x-AFHDS-2A-Transmitter-p-1183313.html?rmmds=detail-left-viewhistory__4&cur_warehouse=CN

Video:

https://youtu.be/EKwYNjRTPWk

hexfet, 2018-02-06:
The description above leaves out one piece of information that was in Antennenfreund's Deviation forum post:  "iRangeX iRX-IR8M 2.4G 8CH Multiprotokoll"

I've used Frsky (D4R-II) and FrskyX (XSR) with both 4-in-1 module in Devo 10 and T8SG without problems.  No other users have reported these range issues to my knowledge.

F-D-R, 2018-02-10:
The iRangeX has quality control issues, there are more people reporting issues with the 4-in-1 module.
BTW you didn't say what power setting you used...

TheRealMoeder, 2018-05-05:
Probably irange Hardware related, so I'm closing this.

closed 2018-05-05 by TheRealMoeder