DeviationTX/deviation #394: T8S - JR bay - crossfire

Flaykz, 2018-04-10:
Hi,

It is plan to support crossfire on T8S jumper thanks to the JR bay ?

Thank you, nice work.

hexfet, 2018-04-21:
Yes, work is underway.  Follow the [discussion here.](https://www.deviationtx.com/forum/protocol-development/7474-serial-out-for-lrs-boxes)

hexfet, 2018-04-27:
Test build (5026091) is [url=https://www.deviationtx.com/downloads-new/category/983-hexfet-protocol-crsf]available[/url] . This build only supports the RC channel values. No telemetry or configuration yet. No Crossfire module to test with but output looks good on analyzer.

On the T8SG-Plus connect the top and bottom JR pins together.  I confirmed with TBS support that if the CRSF signal is present on both pins their module will accept the CRSF protocol on the bottom pin.  The CRSF protocol signal will also appear on the tip connection of the trainer port jack on all transmitters.

To use select the CRSF protocol.  Please report results.

hexfet, 2018-10-01:
The CRSF protocol is confirmed to be working with the TBS Micro module.  However the top JR pin must be disconnected from the JR module, contrary to what TBS Support had said.  See [this post for details](https://www.deviationtx.com/forum/protocol-development/7474-serial-out-for-lrs-boxes?start=140#71049).

closed 2018-11-07 by TheRealMoeder