DeviationTX/deviation #401: www.deviationtx.com down

TheRealMoeder, 2018-05-07:
It gives a 503 when trying to open the site.

F-D-R, 2018-05-07:
I've already tried to contact PB...

PhracturedBlue, 2018-05-07:
I am aware and have filed a ticket to the host about it.

PhracturedBlue, 2018-05-07:
Apparently there is an issue with the PHP version we were using on the hosting provider.  I updated to a later version, and everything appears to be working again.  Please let me know if you see any issues

F-D-R, 2018-05-08:
The Wiki pages still give some warnings:
https://www.deviationtx.com/wiki/supported_models

PhracturedBlue, 2018-05-08:
I fixed it now.  I noticed that there is no 'login' link on the wiki (if you aren't logged in).  Do you know if it was ever there?  I thought there was a 'login' link on the wiki.

PhracturedBlue, 2018-05-08:
It looks like there were some API changes in the latest Wiki code.  I've already fixed a couple of issues, but if you see any PJHP errors on the Wiki, please let me know.

F-D-R, 2018-05-10:
That's because the Wiki isn't within Joomla, only was made to look similar, and only the main menu was ported, so the other plugins and modules, like the login don't work...

TheRealMoeder, 2018-05-12:
I don't know if it is related, but I have problems uploading test builds using the provided perl script...can anyone check?

PhracturedBlue, 2018-05-12:
Yep...the new PHP caught a bug in the uploader script.  Please verify it is wokring now

TheRealMoeder, 2018-05-12:
Works as a treat, once again thanks for the quick fix!

captainju, 2018-08-20:
I don't know if it is related, but some model icons doesn't show up in the forum https://www.deviationtx.com/forum/model-icons
For example here:
https://www.deviationtx.com/forum/bw-icons-for-devo-7e-10/7907-velocidrone-icon-for-7e-10-12e

rebel008, 2019-07-30:
it appears the site is down again as of today(30 jul 2019) any one know anything or able to help?
when trying to goto the site i get; Error: Call to undefined function Joomla\Registry\Format\json_decode(): Call to undefined function Joomla\Registry\Format\json_decode()

TheRealMoeder, 2019-07-30:
@PhracturedBlue @F-D-R can you have a look?

PhracturedBlue, 2019-07-30:
I'll take a look

PhracturedBlue, 2019-07-30:
Apparently the php installation was changed by the hosting provider, though I don't know why.  Switched to newer PHP version, and everything works again

closed 2019-07-30 by PhracturedBlue

TheRealMoeder, 2019-07-30:
Thanks. Forum is still not working

PhracturedBlue, 2019-07-30:
very odd.  I did something and it seems ok now.  I have no idea what happened there

rebel008, 2019-07-31:
you guys are absolutely amazing! big thanks!!!