DeviationTX/deviation #414: Trim in Elevon Mode

Buddywoof, 2018-06-03:
Hi,
I am using a T8SG in elevon mixing mode and I observe that the trim buttons only move one of the control surfaces instead of both.  For elevon mixed mode, the trim should mix using the elevon missing settings, ie, trim should move the control surfaces simultaneously.

Thank you.

vladislavy, 2018-06-04:
Something wrong with your configuration. Just tested my old elevon template and both trims move both control surfaces, in one direction for elevator and in opposite directions for aileron.
https://www.deviationtx.com/forum/model-templates/1308-elevon-and-v-tail-templates-with-dual-rate

Buddywoof, 2018-06-04:
Thanks.   I'll try one of the templates in your link. Which template is best for a 3 channel flying wing with elevon mixing?   Thank you.

Buddywoof, 2018-06-05:
Hi Vladislavy,
I tried the template in your link but getting very strange results.  I see that the template has Ch1=complex and ch2=complex.  However, when I bind the TX with my AFHDS-2A receiver, ch2 changes to "simple".  Any idea what is going on?

Attached is my original elevon mix.  With this mix, elevator or aileron trim only moves one control surface at a time instead of "mixing" and moving both control surface.

[model7.zip](https://github.com/DeviationTX/deviation/files/2070556/model7.zip)

Buddywoof, 2018-06-05:
I figured it out.  On my template, I saw an entry called "usetrim=0".  I changed that to "usetrim=1" and trim is now working perfectly.

PS - I could not figure out where in the T8SG GUI I can set "usetrim" so I just edited the .ini file.

vladislavy, 2018-06-05:
On the complex mixer page at the bottom you can find option "Trim" or "No Trim", press ENT to change.

Buddywoof, 2018-06-05:
That setting seems to toggle between the trim buttons being enabled or not enabled.   "Usetrim" setting in .ini file seems to toggle between single control surface or multiple control surface movement.  I'll have to experiment a bit more but thanks for helping me figure out how to fix the issue. Really appreciate it!

closed 2018-06-07 by vladislavy