DeviationTX/deviation #415: FrSky telemetry from flight controller doesn’t show up with the latest R-XSR firmware

andrejpodzimek, 2018-06-21:
With the [latest F.Port firmware](https://www.frsky-rc.com/wp-content/uploads/Downloads/Firmware/R-XSR/Beta/R-XSR_FPORT_180615.zip) for R-XSR I only get RSSI, but [not the telemetry values from the FC](https://github.com/betaflight/betaflight/issues/6145) (altitude, tension, current). Downgrading the R-XSR firmware to its February version restores law and order and I get all the telemetry data on my Devo 10 with a [“standard” Banggood 4-in-1](https://www.banggood.com/CC2500-NRF24L01-A7105-CYRF6936-4-In-1-RF-Module-For-Walkera-D7E-D10-Transmitter-p-1096707.html). (I’m using LBT.)

My first guess was that something must have changed in the F.Port implementation and therefore the RX doesn’t receive telemetry from the FC any more. It seemed plausible, until it turned out that Betaflight developers cannot reproduce the problem with their equipment; FrSky telemetry just works for them on OpenTX, no matter which receiver firmware version they use.

There must have been a change (buggy or intentional) that made the latest R-XSR firmware (at least the F.Port version) incompatible with Deviation in terms of telemetry.

hexfet, 2018-06-28:
There is a report of successfully using f.port with r-xsr [here](https://www.rcgroups.com/forums/showpost.php?p=39824952&postcount=1233), but the fw version is not stated.  I only have an xsr to test with and its f.port firmware is dated April (I've not yet tested it).

Does the latest non-EU firmware work as expected?

hexfet, 2018-08-11:
Another user reports successfully using deviation with f.port and r-xsr.

From [deviation forum thread](https://www.deviationtx.com/forum/protocol-development/5356-new-frskyx-protocol?start=940#70215):
"Just got it sorted. It seems that the cc3d has pull up resistors on the flexi port uart, and that was disturbing fport telemetey. I removed the pullup and telemetey started working."

closed 2018-08-11 by hexfet