DeviationTX/deviation #418: 繁體中文 Translation item error

Chen-Leon, 2018-07-21:
繁體中文 Translation item error

[image]

TheRealMoeder, 2018-09-03:
Could you please elaborate what specifically is wrong?

goebish, 2018-11-20:
That's weird because this string shouldn't be translated:
https://github.com/DeviationTX/deviation/blob/master/src/protocol/h8_3d_nrf24l01.c#L63
(no _tr_noop for format names)

howard0su, 2018-11-21:
> That's weird because this string shouldn't be translated:
> https://github.com/DeviationTX/deviation/blob/master/src/protocol/h8_3d_nrf24l01.c#L63
> (no _tr_noop for format names)

it is not about this line. the text in red rectangle is the following:
:Model needs reset\nfor standard mixer