DeviationTX/deviation #424: Flysky telemetry - the second temperature and maybe altitude?

Tygrys-1, 2018-09-02:
Could the second temperature sensor be allowed, as it is a place in the telemetry screen for two temperatures? The code only reads the first one.

There is also a Turnigy TGY-CAT01 [Altitude Sensor](https://hobbyking.com/en_us/attitude-sensor-for-i10.html?___store=en_us) . It seems to have an implementation on FS-i6X transmitter telemetry. As I don't have one yet I can not tell for sure how it works, but it will be a nice completion for the Flysky telemetry.

goebish, 2018-09-02:
I think I can add the second temperature sensor telemetry.
Regarding the altitude sensor I'd need a dump because I've no idea of its ID / range, unit ...

Tygrys-1, 2018-09-02:
I understand. I'll try to figure it out if I get one. There is also some magic needed to set/reset the 0 altitude on transmitter side, so maybe we can live without it for some time :-)

You may close the issue on this.

goebish, 2018-09-02:
I think I just found what we're looking for in betaglight source code:
https://github.com/betaflight/betaflight/blob/master/src/main/telemetry/ibus_shared.h#L69
Seems altitude sensor ID is 0x83 and the value is signed centimeter ... please get back to me when you've one of those sensor, I'll make a test build.

Tygrys-1, 2018-09-02:
Maybe take a look at the https://github.com/benb0jangles/FlySky-i6-Mod-/tree/master/10ch%20Mavlink%20Mod%20i6%20Updater

goebish, 2018-09-02:
Yup, same thing:
https://github.com/qba667/MAVLinkToIbus/blob/master/MAVLinkToIBUS/IBUSTelemetry.h#L71

Tygrys-1, 2018-09-02:
Yeah, it is a great way to go. I'll be happy to heve it on the transmitter with alarms, logging, and all the stuff Deviationtx offers.

Tygrys-1, 2018-09-02:
Just be aware that the TGY-CAT01 is a 32-bit sensor data which is not handled by the current code.

goebish, 2018-09-02:
Yeah, I won't implement it unless someone can check if it works anyway.

Tygrys-1, 2018-09-02:
Can you think about GPS and doing it to the INAV/ Betaflight  telemetry code on Ibus, just as the Mavlink example mod?

goebish, 2018-09-02:
Ok, I won't be able to do that right now but it should be fine in October.
Also, I realize that using a 32 bit sensor or betaflight with an altitude sensor could break the current implementation (only if they're sent before 16 bit sensors, that probably doesn't happen as they've id with higher numbers).
Closing this issue for now but it's on my todo list.

PR #430 (2018-09-03): AFHDS 2A: Allow LQI output on Ch13 and Ch14, from goebish
Allow LQI output on Ch13 and Ch14, fixes #422
Add telemetry for 2nd temperature sensor, fixes #424
closed 2018-09-02 by goebish