DeviationTX/deviation #425: Two stage telemetry alarms

Tygrys-1, 2018-09-02:
As for the aircraft battery voltage I'd lika the two stage warnings:
1) battery is low
2) battery i s critical.
Can it be achived with the Deviationtx?  I know I can set whatever alarm on telemerty data, but i have only one stage of it.

TheRealMoeder, 2018-09-03:
Use two telemetry alarms with the same parameter but other cutoff value?

Tygrys-1, 2018-09-03:
Yes.

howard0su, 2018-12-06:
We can make sure one data point change will only trigger one alarm. And the order of alarm is the priotiy of alarm. When critial trigger fired, we can stop the further processing. Does this sound reasonable?

Tygrys-1, 2018-12-17:
Yes, this sounds good. Stop further processing of the sensor when meet the condition for this sensor.
Can you also look at the #440? It is somehow related.