DeviationTX/deviation #433: Jumper t8sg v2 plus

pgibreeze, 2018-09-05:
Hello there. I'm new here and i hope this is the right place.

When I connect my Jumper t8sg to a buddy box (used different boxes) with an "S cable" or a "Trainer cable" to the aux cable, the Jumper's transmitter settings get wiped and reset to defualts (12.00 voltage warning for example).

This doesn't happen right away but after a while. It seems to be connected to the throttle (stick 1 up and down) because the Jumper also only recognized +- 2 (instead of +- 100) on that channel. and it seems to happen when i configure the throttle in the PPM in settings.

When i turn the radio on it warns of low battery and can't save.  It also starts from "Model 207" when turned on. Nothing I do will save and the only way to rectify this is by flashing.

I assume this has something to do with the power from the buddy box creating a surge of some sort which resets and/or hardlocks somthing in the Jumper.

Thanks for helping!

golddogh, 2019-08-24:
I have the same issue yet :(

golddogh, 2019-08-28:
No :/ i broke something in my t8sg when i put 12V so now it does not work...
Maybe you have a solution to reprogram it without usb ? Serial port on stm32 ? It can go in usb mode or dfu mode but not detected properly, so i think the  bios is okay but something went wrong in the memory.

pgibreeze, 2019-08-28:
Hey all. I got notification for this because my case was referenced. Don’t
know if this will help but My issue happens whenever I attach a trainer
plug into my jumper. After a setting up the slave and using it for a while
the jumper goes on the fritz and all channel inputs have crazy trims and
the like. Like throttle jumps to -132 or something and if I move up the
stick all the way it goes to -116 and different things for all channels. On
start up it says there is voltage error and I can’t save anything.

Only way to fix was to do a complete wipe and reinstall everything.

TheRealMoeder, 2019-08-28:
probably same issue as #919

TheRealMoeder, 2019-08-28:
Any chance to make this reproducible so we can trace the issue?