DeviationTX/deviation #441: Loss of binding on Devo Protocol

Morlacus, 2018-09-25:
I have experienced with my devo 12s a binding loss as I was finalizing the set up of a model.  Each time I put the right stick of the tx full up the binding was lost and the rx led flashed. I changed the alim of the Rx, changed the  Rx, Same Result. I formated my USB drive and reloaded the firmware without change. Finally I reproduced the bug on my Devo 8s  after having loaded on it my model file.
In fact the binding was erratic because I used a fixed ID but when the Rx light flashed some channels were blocked.
After more investigation I found that the binding could be not lost if the value of the scale+ in a channel set up was reduced to a certain value. (in the model file included channel 7 scale + reduced to 167.)

 This could come, from expert advice (FDR) from an  overflow  of the channel value, which makes the packet invalid.

[model19.zip](https://github.com/DeviationTX/deviation/files/2415693/model19.zip)

hexfet, 2018-10-02:
From [this post](https://www.deviationtx.com/forum/6-general-discussions/8057-devo-rx-led-flashing#71132):

I was able to duplicate the flashing led behavior with Devo10 nightly build and RX1002 receiver. The receiver is not losing binding and the packets are fine. The PWM outputs are stable and correct even when the LED flashes.

The behavior is triggered by short PWM outputs - less than about 920 microseconds. It only happens if a couple of channels have short output pulses. This makes me think the receiver is mis-interpreting the short output pulses as indicating some type of error condition, and blinking the LED as an indicator.

The PWM output is stable down to at least 900 microseconds on the RX1002. I'm hesitant to change the signal limit in the code, which is mentioned in the doc/Devo.txt file: "The sticks return '0' when centered and seem to max out at 0x640 at either end the upper-nibble of byte 9 indicates whether the channel value is positive/or negative I.e at the end of stick travel the value is 0x640 and bit 9.x indicates which end the stick is at".

I propose addressing the github issue by adding a note to the manual about this behavior.

xTop-SecreTx, 2023-01-30:
Any news about this? I encountered the same problem with blinking RX LED but don't know anything about PWM outputs or something. For me it happens when i want to add more than 7 channels.

hexfet, 2023-02-02:
Well it's not the same problem because this issue was about Devo protocol and SPM4650 looks to be DSMX.

If you haven't already done the freq-fine adjustment procedure, do that.  Go to protocol options and move freq-fine up and down until bind is lost, then set it to a value in the middle.

If you still have issues try changing the other protocol options.  Some receivers have had issues with those settings.  I don't know about that particular receiver but you could search on the forum at deviationtx.com.