DeviationTX/deviation #442: F12E-XMS (5.0.0-21674bd) Crash (use FPV)

unmac, 2018-09-26:
Irregular Crash occurs when F12E-XMS (5.0.0). using FPV vrx.
errors.txt 1
'
[Soft fault]
devof12e-XMS-v5.0.0-21674bd
R0 = 00000045
R1 = 2000bf37
R2 = c0000000
R3 = 00000003
R12 = 2000bf30
LR [R14] (subroutine call return address) = 08009b15
PC [R15] (program counter) = 08009916
PSR = 21000000
BFAR = e000ed38
CFSR = 00000000
HFSR = 00000000
DFSR = 00000000
AFSR = 0000000`

errors.txt 2
'
[Soft fault]
devof12e-XMS-v5.0.0-21674bd
R0 = 00000045
R1 = 2000bf57
R2 = c0000000
R3 = 00000003
R12 = 2000bf48
LR [R14] (subroutine call return address) = 08009b15
PC [R15] (program counter) = 08009914
PSR = 21000000
BFAR = e000ed38
CFSR = 00000000
HFSR = 00000000
DFSR = 00000000
AFSR = 000000'

hardware.ini
'
;Only useful for transmitters with an after-market vibration motor
;enable-haptic=1
;
;switch_types: 3x2, 3x1, 2x2
;extra-switches=
;
;Audio player connected to UART (Devo12/S internally, others via trainer port): audiofx, dfplayer
;voice=
;
;Enable if audio device is wired to uart5 instead of ppm on devo7e-256
;voice-uart5=1
;
;button_types: trim-all, trim-(left|right)-(up|down|both)
;May occur more than once if necessary.
;extra-buttons=
;
[modules]
; there is no need to enable the cyrf6936 module unless
; it is wired to an alternate port. It is Enabled automatically otherwise
; enable-cyrf6936 = B12
; has_pa-cyrf6936 = 1
enable-a7105    = A13
has_pa-a7105    = 1
enable-cc2500   = B3
has_pa-cc2500   = 1
enable-nrf24l01 = A14
has_pa-nrf24l01 = 1
; enable-multimod = A13
'

tx.ini
'
current_model=9
language=0
music_shutdown=1
mode=2
brightness=10
contrast=10
volume=10
audio_vol=10
vibration=1
power_alarm=5
batt_alarm=6450
batt_critical=3900
batt_warning_interval=0
splash_delay=35
[calibrate1]
  max=3665
  min=247
  zero=1948
[calibrate2]
  max=3808
  min=470
  zero=2101
[calibrate3]
  max=3744
  min=458
  zero=2122
[calibrate4]
  max=3650
  min=265
  zero=1955
[calibrate5]
  max=4078
  min=0
  zero=2040
[calibrate6]
  max=3884
  min=0
  zero=2033
[calibrate7]
  max=4095
  min=0
  zero=2016
[calibrate8]
  max=4095
  min=0
  zero=1983
[autodimmer]
timer=0
dimvalue=10
[countdowntimersettings]
prealerttime=0
prealertinterval=0
timeupinterval=0
[telemetry]
temp=celcius
length=meters
alertinterval=1

unmac, 2019-01-10:
Some source code checks.
I suspect that the crash is caused by a watchdog.
I did not confirm the details.

In the TW8816_GetVideoStandard () or TW8816_SetVideoStandard () function
Delay occurrence is presumed to be the cause of watchdog.
(I2C approach ??)

I do not know if this symptom is only my TX.

Do not call VIDEO_GetStandard / VIDEO_SetStandard periodically in the VIDEO_Update () function, 
(to be call only when the value of videosrc changes), the   symptom does not occur.

There are three things that I would like to check further.
  - Adjustment of I2C timeout (LONG_TIMEOUT 0x10000, FLAG_TIMEOUT 0x1000)
  - reconfirmation of connection between I2C and LCD
  - Identify the location where the delay occurs

howard0su, 2019-01-10:
the analyze make sense to me. I will take a look.

howard0su, 2019-01-11:
unfortunately, GetStandard will not immediately return the correct value. We need defer the get/set standard after couple of seconds. will change this.

howard0su, 2019-01-12:
Can u test the build? 
https://www.deviationtx.com/downloads-new/category/1522-csujun-test-video

I don't have seeing such problem in my f12e. Please report back the result.

unmac, 2019-01-13:
ok. i will test the firmware.
I will report the result later.

unmac, 2019-01-14:
After about 2.5 hours of testing,
The symptom did not occur. (firmware : devof12e-XMS-v5.0.0-b360129)

I will use the test firmware few days

howard0su, 2019-01-15:
Thank you for taking time and risk to test.

What's the original repro frequency of the issue? How confidence level you have that it is solved based on this 2.5 hours?

unmac, 2019-01-15:
Symptoms occurred in about 5 minutes in the short case and in about 30 minutes in the long case.
(Within 7~8 minutes of batterys 3 pack at runtime)

It was operating normally while testing 9 batteries 2 time
(9 * 2 = 18).
Keeps TX on while drone-battery is charging

howard0su, 2019-01-17:
I commit the change and you can get it from nightly build then.

closed 2019-01-17 by howard0su

unmac, 2019-01-17:
Thank you. howard0su