DeviationTX/deviation #446: Filesystem corruption on Devo 12E

andrejpodzimek, 2018-10-08:
I’m filing this to (hopefully) get [this old thread](https://www.deviationtx.com/forum/6-general-discussions/7239-is-devo-12e-supported-by-the-deviation-uploader) more attention. There are (AFAICT) two severe issues with Deviation on Devo 12E. I’m not sure whether they are related or not.

1. The Deviation uploader doesn’t support Devo 12E; one has to use an old version of the Dfuse tool to flash it.
2. After flashing Deviation with Dfuse and [creating / restoring the filesystem](https://www.deviationtx.com/forum/6-general-discussions/7239-is-devo-12e-supported-by-the-deviation-uploader#63492), any filesystem accesses over USB other than reads clobber the filesystem so that Devo 12E won’t boot any more. Subsequent attempts to mount the filesystem show ugly binary randomness where file names used to be. Writing the files once, all in one go, onto a freshly formatted FAT12 and without any subsequent modifications, [seems to work just fine](https://www.deviationtx.com/forum/6-general-discussions/7239-is-devo-12e-supported-by-the-deviation-uploader#69707) though.

The obvious problem with the filesystem inconsistency is that (1) it makes it much harder to edit / duplicate your model files over USB and (2) it leaves one wondering whether the problem only affects USB accesses or whether model editing directly on the radio could (theoretically) also be affected.

I also run Deviation on a Devo 10 and it’s just rock solid stable, **no** filesystem corruption ever. So either I have a flawed Devo 12E specimen or there’s indeed a data consistency issue on Devo 12E.

driftbeater, 2019-04-13:
I can confirm that there are problems with the filesystem on Devo 12E. I copied a model setting from a Devo 10 to a Devo 12E, after some flights the settings were reverted to the original version. I now reformatted the USB drive and will from now on only change settings directly on the radio.

richardclli, 2020-12-15:
Same problem with Devo 12S, filesystem will crash occasionally.