DeviationTX/deviation #452: T8SG V2 datalog signature

Tygrys-1, 2018-10-20:
Currently it is the same as the Devo 7e, (0x7e) which is wrong, as the Devo 7e has different datalog entries for buttons and no aux channels. It should be changed to whatever unique for T8SG V2.

Tygrys-1, 2018-12-17:
It will also help to know the telemetry type (Spectrum, Walkera or FrSky) as the datalog file can have different models logged and there is no way to tell the meaning(headers) of the data fields when without this information. 
As there are only few radio type identifiers used, maybe this information can be added to the radio-type data byte, to not change the log structure.

howard0su, 2019-03-01:
All t8 serial changes to 0x18 as this

closed 2019-03-01 by howard0su