DeviationTX/deviation #458: T8SG V2 startup screen.

phaseIV, 2018-11-05:
I’ve installed the latest (nightly) firmware on my T8SG v2 controller and evertything is working super fine! Many thanks for that. There’s just one issue that I’d like to report here; each time I turn on the T8SG there is a popup showing “Ch3 is -6%, safe value = -100”. This message is blocking the controllers operation, I have to press enter to get rid of the popup before the controller starts working.

Is this intentional? Or can I change the configuration so that the controller starts working immediately without the popup? Hope you can help me. Many thanks.

TheRealMoeder, 2018-11-05:
It is intentional, please read this chapter in the manual:
https://www.deviationtx.com/manuals/html-devo10-en/ch_mainpage.html#safety-system

closed 2018-11-05 by TheRealMoeder

phaseIV, 2018-11-05:
Completely missed that. Thanks for the info.