DeviationTX/deviation #470: Nightly build fails after adding parallel builds

TheRealMoeder, 2018-11-29:
See Travis log, there seems to be an issue when running make zip

goebish, 2018-11-29:
I relaunched it and it passed.
I suppose this is because of make -j2

TheRealMoeder, 2018-11-29:
OK. Let's wait for the next nightly before closing the issue.

howard0su, 2018-11-30:
i will take a look

howard0su, 2018-11-30:
where can I find the error? I didn't find error build from: https://travis-ci.org/DeviationTX/deviation/builds

hexfet, 2018-12-01:
I've been using dmake -j4 zips for local test builds.  Sometimes it works and sometimes not.  At least one problem seems to happen at the beginning when parallel jobs are trying to build the cm3 library at the same time. Maybe they're all putting the library in the same destination?

`  CC      sync.c
  CC      dwt.c
  CC      dwt.c
  AR      libopencm3_stm32f1.a
  CC      sync.c
arm-none-eabi-ar: /git/deviation/src/libopencm3/lib/libopencm3_stm32f1.a: Error reading sync.o: File truncated
make[3]: *** [/git/deviation/src/libopencm3/lib/libopencm3_stm32f1.a] Error 1
make[2]: *** [lib/stm32/f1] Error 2
make[1]: *** [libopencm3/lib/libopencm3_stm32f1.a] Error 2
make: *** [zip_devo7e] Error 2
make: *** Waiting for unfinished jobs....
  CC      dwt.c
  AR      libopencm3_stm32f1.a
`

howard0su, 2018-12-02:
the cm3 lib only build once and place the fixed place in current Makefile. I can fix it by move it to individual objs/<tx> folder. however it need change Makefile of libopencm3, which is not good practice.

In Makefile of libopencme, the target folder is not configurable as far as my read:
SRCLIBDIR:= $(realpath lib)
it is fixed to lib.

hexfet, 2018-12-19:
It's a race condition.  On my VM build box it still fails occasionally.  With google's help came up with the following solution. Change lines 69-70 of target/common/devo/Makefile.inc to:
```
$(LIBOPENCM3):
    +flock $(LIBOPENCM3).lock $(MAKE) -C $(SDIR)/libopencm3 TARGETS=stm32/f1 lib; rm -f $(LIBOPENCM3).lock
```
This forces only one of the make jobs to build the libopencm3 library.  That's the process that gets the lock.  The others wait till the library is built.  When the library is built the lock is released.  The waiting jobs still run the libopencm3 make, but the library is already up-to-date so it immediately returns.

But then, in my environment, the build still fails because the optimize.ld file is shared for all targets.  Need to fix that as well...

howard0su, 2018-12-19:
please check PR #493

hexfet, 2018-12-20:
Fixed by #493

closed 2018-12-20 by hexfet