DeviationTX/deviation #479: Need analyze stack usage

howard0su, 2018-12-10:
Here is the report generate by https://dlbeer.co.nz/oss/avstack.html. the following table shows the stack usage based on call graph. Please check the page for the detailed explanation for the following table. I am using f12e as example (7e shows similar functions in the top).

Consider some optimization for the top functions.

  Func                               Cost    Frame   Height
------------------------------------------------------------------------
> reorder_return_cb@objs/devof12e/mixer_page.o     5640     2824        4
> press_cb@objs/devof12e/model_loadsave.o     3504       24       10
> file_press_cb                      3504        8       10
  CONFIG_ResetModel                  3496       16        6
> main                               3496       16       12
> CONFIG_ReadTemplateByIndex         3496       24        7
> usb_cb                             3488        8       10
> invalid_stdmixer_cb                3488       16        6
  CONFIG_ReadModel                   3480       56        5
> protoselect_cb                     3472       48        5
  CONFIG_ReadTemplate                3472       48        5
  RemapChannelsForProtocol           3424      608        4
> mixermode_cb                       3016       48        6
  STDMIXER_SaveSwitches              2968        8        5
  save_switch                        2960      144        4
> set_holdstate_cb                   2952      136        4
> gyro_output_cb                     2864       48        4
> action_cb@objs/devof12e/mixer_curves.o     2840       16        5
> action_cb@objs/devof12e/mixer_setup.o     2832        8        5
> okcancel_cb@objs/devof12e/mixer_setup.o     2824        8        4
  PAGE_SaveMixerSetup                2824        8        4
  MIXER_SetMixers                    2816       24        3
  fix_mixer_dependencies             2792     2768        2
> debug_spi_flash                    1064     1032        4
> PAGE_MainInit                       640       40       13
> limit_cb                            624      456        6
  TGLICO_LoadFonts                    600       40       11
  show_elements                       600       80       12
R _GUI_RefreshScreen                  592       32       11
  EventLoop                           576        8       11
  BATTERY_Check                       568       24       10
> show_iconsel_page                   560       88       11
R GUI_DrawObject                      560        8       10
  LCD_LoadFont                        560       32        9
> GUI_DrawImageHelper                 560       32        9
> LCD_DrawImageFromFile               552       24        9
> _changename_done_cb@objs/devof12e/model_page.o      552        8       10
  GUI_DrawImage                       552       24        9
  CONFIG_SaveModelIfNeeded            544        8        9
> newelem_press_cb                    536        8       12
  CONFIG_WriteModel                   536      128        8
> icon_notify_cb                      536       32       11
  create_element                      528       40       11

howard0su, 2018-12-15:
so far we cannot do anything better than compiler unless we move something as global variables.

PhracturedBlue, 2019-01-14:
These are the most expensive functions from a stack usage perspective:
ini_parse_file                      512      356        6
limit_cb                            668      460        8
RemapChannelsForProtocol           1048      668        4
debug_spi_flash                    1080     1036        5
LCD_DrawWindowedImageFromFile      1248     1036        8
reorder_return_cb@objs/devo8/mixer_page.o     3040     2660        4

I already have PR #583 to address reorder_return_cb

howard0su, 2019-01-14:
debug_spi is not used anywhere.
I am working on LCD_drawwindowedimagefromfile.

PhracturedBlue, 2019-01-14:
it should be easy to reduce the usage of DrawImageFromFile for 7e.  However, as I recall, this is one of the functions whose performance is actually noticeable at the system level.  We do not want to optimize stack-size over performance on this one I think (and if you decide to we'd probably want to do user testing to make sure it isn't noticeable)

howard0su, 2019-01-14:
I think we need fix the function. When I read the code, I realized the
major problem for this function is not stack usage but the risky of stack
overflow. We didn't check if the image width against 480 and blind trust we
will not having large image. I would prefer having a fixed size of buffer
and read amount of pixel data and process them in batches.

In term of performance, I believe we need fix it via having a function
called DrawHLinePattern(x,y, bufffer, length) to be implemented inside LCD
driver for each driver. It will hugely speed up the speed to draw the image.