DeviationTX/deviation #496: [TODO] Cleanup Telemetry Logic

howard0su, 2018-12-19:
The current telemetry has the different set of logic to handle different vendors telemetry. It cause the code is not well formatted and have to do lots of redundant code to handle the same or similar logics.

Someone need analyze the code and come out two common abstractions:
1. A standard set of telemetry values. This set of telemetry values are super set of all vendors telemetry values. the vendors telemetries can be mapped into this super set. 
each values has the predefined name, unit and range. All vendors telemetry have to be converted into this. 
when necessary, we can drop some vendor's special telemetry to simplify the implementation. Or we can have some extension API to support some special telemetries.

2. A standard set of decoding functions to decode the stream and convert to the right range and right unit. 
I can see lots of duplicated code to decode the stream. this API can be either functions or just a set of MACROs to make decoding code much clean for each vendors.

TheRealMoeder, 2018-12-19:
I fully agree with your observation, there is a lot of ROM wasted in telemetry. Whatever we do though, we should be again very careful not to brake existing model files.

howard0su, 2018-12-19:
I think keep back compatibility is hard as today's name is not consistent. for example: 
telem_dsm.c:        case TELEM_DSM_FLOG_RSSI_DBM:   strcpy(str, _tr("RSSI")); break;
telem_crsf.c:    case TELEM_CRSF_RX_RSSI2: sprintf(str, "rxRSSI%d", telem-TELEM_CRSF_RX_RSSI1+1); break;

maybe another PC side script to convert. it can make the life simpler.

howard0su, 2018-12-19:
maybe time to think about 6.0 BRANCHING.