DeviationTX/deviation #506: No Chinese manual is built

howard0su, 2018-12-20:
There is not CHN manual listed here:
https://www.deviationtx.com/user-manual

Can u build one? I am working on transifex site to input the Chinese translations. some chapters are finished and ready for more broad access.

TheRealMoeder, 2018-12-20:
The HTML manual is already built and uploaded with daily updates pulled from transifex. You can access it [here for devo7/10](https://www.deviationtx.com/manuals/html-devo10-zh/). @F-D-R has yet to set the link though. I will enable pdf build in makefile.

TheRealMoeder, 2018-12-20:
I remember now why pdf wasn't enabled for Chinese: sphinx doesn't support it.

howard0su, 2018-12-20:
Your link works great. I can see the changes I made today. Let's add the webpage first.

howard0su, 2018-12-24:
I found some link talking about workaround to support Chinese:

https://translate.google.com/translate?sl=auto&tl=en&u=https%3A%2F%2Fwww.cnblogs.com%2Fwindtail%2Fp%2F8289421.html

TheRealMoeder, 2018-12-29:
I tried getting a chinese pdf output by switching to xetex, but the current sphinx version does not support cyrillic with xetex as latex-engine right now, so this would actually be a step backwards with loosing the russian PDF. This behavior will change in Sphinx 2.0.0, so we might look into getting a chinese PDF manual when this version it is released.

howard0su, 2018-12-30:
Let's wait for 2.0 then.

PhracturedBlue, 2019-01-22:
I can't build the pdf, but the html manual is available on the website now

closed 2019-01-22 by PhracturedBlue