DeviationTX/deviation #536: Optimize Mixer code path

howard0su, 2019-01-04:
I did a gcov data for mixer code path, here is some finding. @hexfet do you have time to optimize this?

I execute MIXER_CalcChannels 50,000 times.
`
   500000:   72:void MIXER_EvalMixers(volatile s32 *raw)
        -:   73:{
        -:   74:    int i;
        -:   75:    s32 orig_value[NUM_CHANNELS];
        -:   76:    //3rd step: apply mixers
 13500000:   77:    for (i = 0; i < NUM_CHANNELS; i++) {
^^^^ We should use the real number of channels instead of possible channels.
 13000000:   78:        orig_value[i] = raw[i + NUM_INPUTS + 1];
        -:   79:    }
   500000:   80:    for (i = 0; i < NUM_MIXERS; i++) {
   500000:   81:        struct Mixer *mixer = &Model.mixers[i];
        -:   82:        // Linkers are pre-ordred such that we can process them in order
   500000:   83:        if (MIXER_SRC(mixer->src) == 0) {
        -:   84:            // Mixer is not defined so we are done
   500000:   85:            break;
        -:   86:        }
        -:   87:        //apply_mixer updates mixed[mixer->dest]
    #####:   88:        MIXER_ApplyMixer(mixer, raw, &orig_value[mixer->dest]);
        -:   89:    }
        -:   90:
   500000:   91:}
`

howard0su, 2019-01-04:
> I did a gcov data for mixer code path, here is some finding. @hexfet do you have time to optimize this?
>
> I execute MIXER_CalcChannels 50,000 times.
>
> 500000: 72:void MIXER_EvalMixers(volatile s32 [...]

howard0su, 2019-01-04:
This function need build a mapping table to speed up. the execution time is 165x.

 89000000:   93:unsigned MIXER_MapChannel(unsigned channel)
        -:   94:{
 89000000:   95:    switch(Transmitter.mode) {
        -:   96:    case MODE_1:
    #####:   97:       return channel;
        -:   98:    case MODE_2:
    #####:   99:       switch(channel) {
    #####:  100:       case INP_THROTTLE: return INP_ELEVATOR;
    #####:  101:       case INP_ELEVATOR: return INP_THROTTLE;
    #####:  102:       default: return channel;
        -:  103:       }
        -:  104:       break;
        -:  105:    case MODE_3:
    #####:  106:       switch(channel) {
    #####:  107:       case INP_AILERON:  return INP_RUDDER;
    #####:  108:       case INP_RUDDER:   return INP_AILERON;
    #####:  109:       default: return channel;
        -:  110:       }
        -:  111:       break;
        -:  112:    case MODE_4:
    #####:  113:       switch(channel) {
    #####:  114:       case INP_AILERON:  return INP_RUDDER;
    #####:  115:       case INP_THROTTLE: return INP_ELEVATOR;
    #####:  116:       case INP_ELEVATOR: return INP_THROTTLE;
    #####:  117:       case INP_RUDDER:   return INP_AILERON;
    #####:  118:       default: return channel;
        -:  119:       }
        -:  120:       break;
        -:  121:    }
 89000000:  122:    return channel;

howard0su, 2019-01-04:
this function is not optimized.

  8000000:  474:s32 get_trim(unsigned src)
        -:  475:{
        -:  476:    int i;
 88000000:  477:    for (i = 0; i < NUM_TRIMS; i++) {
 80000000:  478:        if (MIXER_MapChannel(Model.trims[i].src) == src) {
    #####:  479:            return MIXER_GetTrimValue(i);
        -:  480:        }
        -:  481:    }
  8000000:  482:    return 0;
        -:  483:}

PhracturedBlue, 2019-01-04:
before taking on too much change here, I'd like to see the unit test harness integrated and tests written for the mixer.  This is the core of deviation, and I know I've screwed it up enough times trying to alter it.

howard0su, 2019-01-07:
Add a simple profiler, maybe someone can collect some stats.

https://github.com/howard0su/deviation/commit/78d04100766b