DeviationTX/deviation #541: Consistent Naming conversation

howard0su, 2019-01-05:
The following is a sample definition list. All the macros start with HAS but actual they are referring two different things. One is hardware confirmation, other is software feature set.
#define HAS_ADVANCED_GUI    1
#define HAS_PERMANENT_TIMER 0
#define HAS_TELEMETRY       1
#define HAS_EXTENDED_TELEMETRY 0
#define HAS_TOUCH           0
#define HAS_RTC             0
#define HAS_VIBRATINGMOTOR  1
#define HAS_DATALOG         0
#define HAS_SCANNER         0
#define HAS_LAYOUT_EDITOR   0
#define HAS_EXTRA_SWITCHES  OPTIONAL
#define HAS_EXTRA_BUTTONS   0
#define HAS_MULTIMOD_SUPPORT 1
#define HAS_VIDEO           0
#define HAS_4IN1_FLASH      0
#define HAS_EXTENDED_AUDIO  0
#define HAS_AUDIO_UART5     0
#define HAS_MUSIC_CONFIG    0

I would propose change all software feature switch with the prefix "SUPPORT_" and leave HAS_ as the prefix to define the hardware capability.

This is one step further to support building a website to let user customize firmware by select the software features.

Thoughts?

PhracturedBlue, 2019-01-05:
I'm in favor.  Before going there though, I'd recommend documenting each of these defines.  I hate having to use #ifdefs in general, and having a documented list of them may have the side-effect of being able to remove/consolidate some of them.

It also gets a little weird with the 4_IN_1 and MUTLIMOD.  This is a software flag that enables searching for a hardware addition.  I understand your classification, but 'SUPPORT_MULTIMOD_SUPPORT' is not a very good name...

howard0su, 2019-01-05:
Where is the place to write dev doc?

I am open for better names/prefix. (i am not good at naming.)

PhracturedBlue, 2019-01-05:
I think in the code will be best.  I'm wondering if we should create a 'default_defs.h' that ensures that every one of these has a value (and then replace all #ifdef with #if, or even better if().
It would give an easy place to find these, and give us the best chance to actually keep the documentation current.

howard0su, 2019-01-05:
okay. I like the idea to use code.

I am actually thinking about the same thing to have a default_defs.h to
make sure the consistency meaning of not-defined macro == macro defined as
0. Also this file can serve as the purpose for the application/user to
parse the available options.