DeviationTX/deviation #553: Help Needed to test Tx with test builds

howard0su, 2019-01-07:
I want to collect who have which Tx can help test certain changes:

- [ ] at9 -- ()
- [ ] at9 -- ()
- [ ] at9 -- ()
- [ ] at9 -- ()
- [ ] at9 -- ()
- [ ] at9 -- ()
- [ ] at9 -- ()
- [ ] at9 -- ()

goebish, 2019-01-07:
devo10 + t8sg + t8sg v2 plus

TheRealMoeder, 2019-01-07:
devo7e, t8sg, t8sg v2+

PhracturedBlue, 2019-01-07:
at9, devo10, devo12, devo8, devo7e (w and w/out 256) devof12e, devof7 (t8sg, t8sg v2 plus, x9d)

But I'm not flying these days, so am limited only to what I can test on the ground.

howard0su, 2019-01-07:
please help verify new compiler and new libopencm3. I got report that combined both change cause infinit restart on devo12. I would like to know the crash so that I can fix.

howard0su, 2019-01-07:
What is the status of x9d? As today the build is broken.

PhracturedBlue, 2019-01-07:
x9d build was never finished because there was no way to support multiple protocols at the time.  That has changed since with the 4-in-1 module I think, so maybe it is worth a re-look.  It is a pretty difficult port as the hardware is very different.  I'm not sure if we'd win over many users given it has a good open-source firmware already

TheRealMoeder, 2019-01-07:
12e shouldn't matter, as it is basically a 10 with two additional knobs

hexfet, 2019-01-07:
devo10, t8sg, t8sg v2 plus

howard0su, 2019-01-08:
Looks like we have reasonable coverage for almost every Tx. Can I request that we can have bi-weekly test activity to test various branch to make sure we can have enough tests for some changes.

The things in my mind is:
compiler change #480 
linker script change #550

caixiaohai0, 2019-01-12:
d8s,d12s,d10,d7e（256）

closed 2019-04-28 by TheRealMoeder