DeviationTX/deviation #574: Calibration is broken in master

howard0su, 2019-01-12:
after calibration, the stick value and swich value will not change anymore. @TheRealMoeder @PhracturedBlue

TheRealMoeder, 2019-01-12:
Cannot reproduce this. Calibration works just fine for t8sg_v2_plus in 37c9f4e

howard0su, 2019-01-12:
i can repro with the exact same build on devof12e. After calibration, the
tx.ini shows followings:
[calibrate1]
  max=1947
  min=1947
  zero=1947
[calibrate2]
  max=3698
  min=3698
  zero=3698
[calibrate3]
  max=2070
  min=2070
  zero=2070
[calibrate4]
  max=1915
  min=1915
  zero=1915
[calibrate5]
  max=4094
  min=4094
  zero=4094
[calibrate6]
  max=3904
  min=3904
  zero=3904

howard0su, 2019-01-13:
bisect shows the break is introduced by this change. It shows on my
devof12e, also I got report from community.

commit 6d419f13ebabd23bc15b80061362a135f26df48e (HEAD, refs/bisect/bad)
Author: hexfet <jon.plotky@gmail.com>
Date:   Tue Jan 8 14:04:14 2019 -0500

    Reduce stick-to-air latency in some protocols with manual mixer
scheduling (#532)

howard0su, 2019-01-13:
Repro steps:
1. flash latest master to devof12e (maybe devo10 as well)
2. go to menu Tx menu, then Tx config, then Calibrate
3. follow the instruction
4. go to channel monitor, you will see sticks change doesn't change the
display of the mixer output.
5. I changed some protocols like devo which is changed by the change, or
wfly which is not changed by the change. both can repro. sometime, the
min=max=0, sometimes min=max=(a large number).

howard0su, 2019-01-13:
I need some help to narrow down. The only code change worries me if the
following ISR behavior is changed? Did we stop this ISR somehow in some
scenarios?

void exti1_isr()
 {
-    //ADC_StartCapture();
-    //ADC completion will trigger update
+    // medium_priority_cb();  Currently not used. If needed,
+    // use exti3 for mixer updates.
     ADC_Filter();
-    medium_priority_cb();
+    MIXER_CalcChannels();
+    if (mixer_sync == MIX_NOT_DONE) mixer_sync = MIX_DONE;
 }

howard0su, 2019-01-13:
If I do this patch, the problem is fixed. Any idea why this happened
@@ -138,6 +140,7 @@ void PAGE_CalibInit(int page)
             GUI_RefreshScreen();
             priority_ready = 0;
         }
+        ADC_Filter();
         for (u8 i = 0; i < INP_HAS_CALIBRATION; i++) {
             s32 value = CHAN_ReadRawInput(i + 1);
             if (value > Transmitter.calibration[i].max)

howard0su, 2019-01-13:
@hexfet can u show me some lights here? Not sure how devof12e is special here after your change.

hexfet, 2019-01-13:
Just noticed this myself.  Calibration works if the currently selected protocol is not one with optimized latency.  I'm guessing the calibration stops the protocol callback.  Will find a fix.

PR #582 (2019-01-14): Fix calibration (issue #574)
Run mixer calculations on timer after de-initializing protocol.  Ensures channel data is updated for functions like calibration.  Fixes #574 .
closed 2019-01-14 by hexfet