DeviationTX/deviation #577: Improve Travis build coverage

PhracturedBlue, 2019-01-12:
Currently our target matrix for travis does not cover all of the targets in the nightly build.  Additionally, the travis build takes almost twice as long to run as the nightly build.  The primary reason for the matrix build was to reduce runtime, and we seem to be failing at that.  We need to rethink the travis build setup

howard0su, 2019-01-12:
I actual don't mind to have a longer runtime for travis. Normally my project in the work take about 45minutes to finish. I would prefer to add all targets into travis-ci.

PhracturedBlue, 2019-01-12:
because I like to have a quick response to a change if it is broken

TheRealMoeder, 2019-01-12:
Matrix is faster, since usually, at least four builds are running in parallel and each is less than two minutes. Nightly takes over nine minutes until we get first results.

TheRealMoeder, 2019-01-12:
Check PR #578

PhracturedBlue, 2019-01-13:
I'm not sure how travis reports build time.  I would expect that the nightly build will fail really quickly if we break devo8 or devo10 since they build 1st, so even the monolithic build may be faster than the matrix build in the fast-fail as well as the full-run case.  And with a monolthic build we could probably increase parallelism and see further benefit (I'm not sure how travis allocates resources, so maybe not).  On the other hand, the logs from monolithic builds are more difficult to parse.

When I've worked with gitlab or Jenkins CI, I am able to setup sophisticated multi-stage builds where you can have a common setup and then parallelize just the build stage.  and treat each part as its own stage.  This could be helpful, because we do not really benefit from having separate VMs for each matrix target, and the setup time is significant.

howard0su, 2019-01-13:
Did u try this? https://docs.travis-ci.com/user/build-stages If not I can take a look today.

TheRealMoeder, 2019-01-13:
From what I've read, each build stage job is run in a separate VM, so this wouldn't give any gain.

TheRealMoeder, 2019-01-13:
How is this going to be faster than current approach with (expanded) matrix? Currently we have first results for major targets within a minute. Also, we have small logs and at a glance you can see which target fails.

howard0su, 2019-01-13:
Current approach, we need go to click the link to travis website to find
out when the job is running. the status only reports back when the job
finish. I just want something which can fail if some early targets already
failed before long running job (make zips).

TheRealMoeder, 2019-01-13:
If I'm not mistaken, if one of the Travis matrix jobs fails, we already get the red x in GitHub for the commit. Build time is around 20 seconds per target, about 40 seconds are needed for setting up the VM before the build begins. So having more than 3 builds in each Travis job will not be anymore efficient. I guess one option would be to make matrix of 5 for the larger targets (each major transmitter/emu line: black and white, color, text, emu, wine) where zips for all "subtargets" are build and the build script posting a GitHub status back for each individual target.

howard0su, 2019-01-13:
if that's the case, my suggestion doesn't matter.

I think your suggestion works. maybe you want another target other than
zips.

howard0su, 2019-01-18:
shall we at least add at9, devof4 into the build matrix?

PhracturedBlue, 2019-01-18:
at9 should already be there today.  I'm not opposed to adding f4

howard0su, 2019-01-18:
yes. why at9 is like TARGET= but others are MAKETARGET. what's the
difference?

PhracturedBlue, 2019-01-18:
there shouldn't be any.  probably just added incorrectly