DeviationTX/deviation #589: Replace Make build system

PhracturedBlue, 2019-01-14:
Our current make files are quite complicated using multiple includes and various overrides conditionals, shell commands, etc.  It is also difficult to programatically define build parameters.  It is extremely difficult to debug when issues occur

There has been mention of replacing the Makefiles in the past.

What alternatives are there?

SCons: Pyton based, should be able to implement our build rules.  I believe it is extremely flexible.  I've never actually used it in the past.  We already require python installation to build Deviation (easy new build dependencies?)

CMake: Cleaner syntax than make.  In my experience it tends to rely heavily on existing macros.  I have mostly seen it used to replace autotools.  Not sure how well it works with a custom build system like Deviation has.  In my limited experience, I have found it difficult to debug in the past when trying to configure a build.

anyone have experience with there or other options?  Other opinions?

howard0su, 2019-01-15:
I had some experience on CMake but didn't work on SCons before.

defr, 2019-01-15:
Mostly been lurking around there, but if you're changing you might want to consider Meson, which is Python based too and has more traction than Scons right now.

howard0su, 2019-01-15:
If I think about the integration with other IDE (even visualstudio), CMake is clearly better than any other build tools. but Scon document is pretty fair:
https://github.com/SCons/scons/wiki/SconsVsOtherBuildTools

PhracturedBlue, 2019-01-16:
Some notes...
I'm not super concerned about integration with an IDE.  All options can integrate with VisualStudio and Eclipse.  I am not a big IDE fan, and am not likely to base a decision on this functionality.

I looked at meson, and I'm not sure it is low-level enough for my tastes.  I don't like abstracting functionality to the build system...it makes it harder to debug and control.  Without the ability to write functions, we're reliant on meson capabilities meeting all our needs, and/or using helper scripts instead of inline code like we have in Makefiles.  I am not convinced this will make maintenance easier.  It would take a deep-dive investigation to determine this.  On the other hand, I do like the generation of ninja files.  I think that would make it easier to figure out what is going on. 

SCons:  This looks viable to me.  It has similar concepts to make, and allows coding rules in python.  I don't believe there would be any issue translating our current code to SCons.  Debuggability seems pretty well supported.  It is supposedly pretty slow and some people won't like that it doesn't have sticky configuration (though to me that is ok)

CMake- supports functions and macros which should make it possible to map our existing functionality.  It has a lot of pre-defined functionality (for things like cross-compiling and dependency handling)  I'm not sure how easy it is to bypass for fine-grain control of the build.  There are some build-script debug options, I can't tell how useful they are.  It can generate ninja files which seems pretty cool.

All of them require extra steps for installation...not much we can do about that.  The python based ones may be a little easier to setup than CMake on windows.

Realistically I may need to actually try to setup a build in each and see what it looks like, and that will take some time.