DeviationTX/deviation #597: Scrollbar is hard to click

howard0su, 2019-01-15:
the scrollbar is close to right edge of screen and only the top button and bottom button is clickable. This make scroll very hard to operate via touch screen.

Suggestions: 
1. make scrollbar a bit wider to make it easy to access
2. Make the bar itself clickable. this is very standard behavior that scrollbar can be click on the bar. if the clock above the scrollbar position, scroll up otherwise scroll down.
3. even better can me support drag? maybe redraw performance will be problem. worth to probe?

PhracturedBlue, 2019-01-16:
clicking above/below the scrollpos should be easy to implement.  dragging would be complicated if you wanted live updates...just moving the scroll position and updating on release is probably not too difficult.  Making it wider potentially requires relayout of pages.  We have some pages that allocate every single pixel in 'x' as I recall.

howard0su, 2019-01-16:
this is comment from community. Will u work on this?

PhracturedBlue, 2019-01-22:
This is should be fixed by #606

closed 2019-01-22 by PhracturedBlue