DeviationTX/deviation #610: UI Loc

howard0su, 2019-01-18:
Shall we use the same site as doc to translate the UI? We are a bit behind on UI translation.

TheRealMoeder, 2019-01-18:
I should be easy to do, but I didn't set it up because by default transifex doesn't understand our language file format. We have to think about what the process here could be. We have to make a script which converts a source and destination file pair into our own Lang format. Then source strings could be regularly extracted by a Travis creator cron job and pushed to Travis. Every nightly run would then pull translations from transifex and should also push those files back to the repo (mainly for backup purposes, in case transifex shuts down)

TheRealMoeder, 2019-01-18:
Oh, you're right.. Then it should be simple to do... The only thing I don't like about my translation solution for deviation-manual, is that we have a daily commit in the repo due to the cron job, even if there are no new translations, caused by the timestamps differing in the translation after being pulled from transifex. While not an issue for the manual really, it would make a nasty commit history in the firmware.

howard0su, 2019-01-18:
then you can differ this import as a manual work. you can do it once a week
or adhoc. nightly build doesn't care it that much but release build should.

howard0su, 2019-01-18:
or you can look at hooks: https://help.github.com/articles/about-webhooks/

PhracturedBlue, 2019-01-18:
1st off.  please don't setup anything that auto-commits to the deviation repo.
2nd off.  can you please give me access to the DeviationTx organization on transifex

We could create a sub-repo that is populated automatically with transifex data.  That could be triggered by the nightly build system or something similar.  I don't want deviation builds to ever be built directly off live data from the website.

The current solution for devation-manual needs to be fixed.  We should never do a commit if nothing has changed but a timestamp.  I'll take a look at fixing it.  I'm of a mind to rewrite the entire history on deviation-manual to get rid of all the useless commits there.  Once we have a proper way to do the merge, we'd use the same method here.

howard0su, 2019-01-18:
what's the plan for the localization?

PhracturedBlue, 2019-01-19:
@TheRealMoeder no need to apologize.  I really like this solution to managing translations.  I have reworked the travis-ci code for the deviation-manual to be something I can live with.

@howard0su I don't follow your question.  I thought that's what we were discussing in this thread.
My thought is the basically the same as yours:
1) Travis generates po files and sends them to transifex
2) we setup a new github project to consume the changes from transifex and auto-commit nightly
3) we add this as a subproject to deviation
4) we modify Make to build the language files from the po files
5) ??? anything I missed

My current concern is how to associate a deviation build with the proper version of the language repo so that builds can be reproduced.  This would be much easier if we didn't use a subrepo, but then I need to be willing to let travis commit back to this repo.

howard0su, 2019-01-19:
This is what I am asking. just like to have the clarification on the plan.

I am complete fine with the proposal. I think subproject is not necessary bad practice.

One small suggestion is that we need somehow promote the translation project to the community. And bring the awareness on the missing translations, like Range Test feature, it is there for a while but without translation as today. A number of users have the passion of changing the translation. I saw the community even maintain its own translation file in Chinese, which I am encouraging them to contribute it back to the repo.

PhracturedBlue, 2019-01-19:
I haven't added the code on the Deviation side to parse this, but I uploaded the current translations to a new project on transifex:
https://www.transifex.com/deviationtx/deviation-1/dashboard/

I split Chinese into:
zh_CN and zh_TW since that is what we seem to have in our translations, please let me know if I should do something different.

howard0su, 2019-01-19:
Can u make sure I am reviewer for zh-CN?

PhracturedBlue, 2019-01-19:
Note that I probably will need to re-generate the seed translations as I work on getting everything working right.  I do not recommend updating translations until all the code is in place

PhracturedBlue, 2019-01-19:
Also, if anyone has any good ideas on how to handle multiple translations for the same string, I'm open to suggestions.
It is a bit odd because the translator can decide on any given string whether to support per-tx translations on a string-by-string basis.  I haven't thought of a good way to do this yet

PhracturedBlue, 2019-01-19:
we could have 3 identical files to translate...that would be easiest, but I think overkill.
I'm currently thinking of doing something similar to what we have now.
For any string, multiple translations are supported as:

| English | Translation |
| -------- | ------------ |
| string1 | translation1 |
| string2 | translation2<br>devo8:alt-translation2<br>devo12:alt2-translation2 |

TheRealMoeder, 2019-01-19:
Also a good solution, if we add a documention for this upfront, so everybody knows about the usage of default strings and for which tx's those subspecifications are valid (e.g. devo8 works for 6 as well), otherwise users maybe confused. I also wonder if it's somehow worth adding (the possibility of) a comment to each line, since sometimes it is helpful knowing in which context the translation is needed.  In the manual, strings are a lot longer and self-explaining...

PhracturedBlue, 2019-01-22:
Fixed by #629 

We still need to figure out how to broadcast this to our users.

closed 2019-01-22 by howard0su