DeviationTX/deviation #617: Revert the new language format support

howard0su, 2019-01-19:
We got the complain from forum on the new language format which break the back-compatibility. I want to start the thread to discuss what we should do:
1. new format has some advantages, compact which saves the disk space, I/O. pre-caculate hash reduce the time to switch a language. also there is disadvantage, break compatibility is one and block user to manually edit is another disadvantage.

2. We need set the clear expectation on the community that nightly can potentially break the existing functionality and may need complete refresh the Tx to recovery. so backup SPI flash is must step.

3. How can we tag the good nightly build for more broad testing and evaluate? we may not want to upload nightly for everyone. instead, we can only publish some LastKnowGood builds (LKG) to public.

4. Shall we start preparing new release?

howard0su, 2019-01-19:
here is the thread:
https://www.deviationtx.com/forum/6-general-discussions/7814-jumper-t8sg-plus-stopped-working-after-fw-upgrade#72356

PhracturedBlue, 2019-01-19:
Why does it break backward compatibility?  I thought we detect the file format and use the right one?
I don't really like the new format personally because it makes it impossible for users to alter their translations (which makes it harder to test language fixes).  But I understand the need for it on tx with small filesystems.  My preference is probably to only enable the new format on small-fs tx, and leave the others with the original format.

I'm not interested in debating questions 2 or 3.  Nightlies are less well tested than releases.  Our goal is to use Travis to try to make nightlies stable, and we can certainly continue to do more, but I don't think I want to create another tier of builds.  These questions would also be mostly resolved if we did more regular releases.

As for doing more regular releases....Releases are quite a lot of work, so doing them frequently requires someone willing to do the work.  It requires testing multiple protocols on multiple transmitters, which means either getting community support or spending developer time to do that.  I'm not really interested in spending my bandwidth at the moment organizing a release.  If someone wants to take on the task, then I'm open to it.  Mostly it means enforcing a code-freeze, getting testing done by users, getting bugs logged and fixed, reviewing and updating documentation, and preparing the release notes.

PhracturedBlue, 2019-01-20:
Using the elffile doesn't work.  The linker optimizes strings, so (for instance) if you have the string '13 Point' and '3 Point', only '13 Point' will appear in the elf file, and the pointer for '3 Point' will just be the same a the pointer for '13 Point'+1

howard0su, 2019-01-20:
i see. this will be a challenge for LTO build then. LTO's object file doesn't exist any real stuff just pseudo code. We cannot parse it to figure out what string is used.

howard0su, 2019-01-20:
oh, forgot lto can support  -ffat-lto-objects. then the object files should contain string table as usual. Need double check though.

howard0su, 2019-01-22:
reverted.

closed 2019-01-22 by howard0su