DeviationTX/deviation #632: Handling Japanese translation

PhracturedBlue, 2019-01-22:
We have a japanese translation now, thanks to transifex and Toshihiro Makuuchi.  However, we don't have a check to verify that all necessary characters in the language files are available in our font files, and the translation is mostly unusable today (from how many strings are blank in the emulator...i can't read Japanese)

We can use `utils/build_unicode_table.pl src/fs/language/locale/*.po` to locate all needed characters, but still need to cross-check against the fonts

howard0su, 2019-01-22:
the font need regenerate for JA. We didn't includes the characters for Japanese.

howard0su, 2019-01-22:
we need a free Japanese font to regenerate bfd files.

commit 59bc2299 (2016-04-30): Fix LCD_Clear function to be reliable on Devo7e.  Fixes #632
closed 2019-02-02 by PhracturedBlue