DeviationTX/deviation #634: Devo8 dev build is broken

howard0su, 2019-01-22:
It is over ROM size limit:
 region `rom' overflowed by 740 bytes

My suggestions:
1. Don't enable printf by default in dev build
2. Enable debuglog in dev build
3. developer can use CFLAGS_clock = -DENABLE_DEBUG_PRINT to enable printf per file.

Thoughts?

PhracturedBlue, 2019-01-22:
Not really my favorite answer.  A lot of the upcoming patches should address this (LTO and/ or your model configuration).  So this issue should resolve itself pretty soon.  It seems like a lot of work to need to manually enable debug in that way.  possibly if we had classes of debug (protocol, gui, interrupt, buttons, etc) that we could choose to enable or not, it'd be better, but that classification is also a lot of work.

howard0su, 2019-02-08:
LTO seems to solve the problem.
ROM: 0x08004000 - 0x0803debc = 231.68kB
RAM: 0x20000000 - 0x200067f8 =  25.99kB

closed 2020-03-14 by vladislavy