DeviationTX/deviation #671: Language Index is not stable

howard0su, 2019-02-07:
During the work on the testing for each page, I noticed a issue that the code assumes the stable of file order in one folder. For example, we use index as language index and assume the lang.* under language is stable and one index always mapped into one file. This maybe true when user didn't reformat the flash. Unfortunately, this will not be true if user copy the new files into the folder like recent change on f12e localization, or user complete reformat the SPI falsh, or during different run of unit testing.

I am thinking to use the filename hash instead of index as the langid. However this may break existing tx.ini and no easy way to maintain the back compatibility.

Any idea or suggestions?

PhracturedBlue, 2019-02-07:
store the filename in the ini.  translate to and from index during model load.  the index will be stable as long as the tx is on, so it is fine to use locally.  we can easily support reading index and writing filename so we can maintain backward compatibility.

howard0su, 2019-02-08:
since our language is always using two characters to represent. I can use u16 instead of u8 to store the language. So that I don't need change lots of code to handle a string as language. sounds good?

PhracturedBlue, 2019-02-08:
use a u32 and store the entire 3 character extension.  it is possible we'll need to use the extra character in the future.  Otherwise I'm ok with it.

PhracturedBlue, 2019-02-08:
But to be clear, make sure the ini file is human readable (it should say 'fr' not '26226')

howard0su, 2019-02-08:
let me wait for config module cleanup done then.

commit 9fd7f5e5 (2016-04-22): Fix scrolling in scrollable placed in a Dialog.  Fixes #671