DeviationTX/deviation #673: [UserVoice] Too hard to select voice in Voice Configure page

howard0su, 2019-02-09:
The current UI requires select a voice via click many times when there are several voice file. User requested if we can make it easy to select like show a list of files.

TheRealMoeder, 2019-02-09:
How would a list be faster? We would still have to scroll down a long way if voice.ini is large. The current implementation allows long press to jump in steps of 10, which I have found to be pretty convenient.

caixiaohai0, 2019-02-10:
The list display allows you to select the desired voice file more directly.

TheRealMoeder, 2019-02-10:
Are we talking a black and white screen or the color GUI here? I agree, on color GUI selecting the voice could be made easier using its larger display and a touchable list of files. But on the b&w screen (devo 7e/10, t8sg etc.) I doubt we would really gain anything. We could display a maximum of 4 files (one per row), and still you would have to click for every file to get to the one you want. As the labels are read on the fly from the filesystem, this might even be slower than the current solution.

caixiaohai0, 2019-02-11:
OMG,This is my mistake, not clearly expressed, I am talking about the color GUI, devo6/8/12，I'm sorry

TheRealMoeder, 2019-02-11:
I've put it on my to-do-list ;)