DeviationTX/deviation #677: Introduce layout to our UI framework

howard0su, 2019-02-11:
there is lots of code to handle the layout of the pages. There is pretty standardized technology used to create self adopted UI for smartphones, which can work across various screen size.

I believe we should adopt such technology as well.

The high level thinking is pretty simple. There are three type of layout:
1. flow layout
2. grid layout
3. customized layout

all the pages today can be considered as customized layout. and we can move or redesign the pages into the first and second layout. This will also remove the different code for 128x64 v.s. 320x240.

reference:
https://material.io/design/layout/responsive-layout-grid.html#grid-customization

PhracturedBlue, 2019-02-11:
I'm not sure it can be done without adding significantly to the code size.  However, given that the number of screen-sizes is fixed, perhaps that logic could be pre-computed.
One issue you'll find is that the 128x64 screen is so low-res that in some cases, we use alternate strings to display information.  There are also cases where we use a fundamentally different widget for the larger screens (using a scroll-list rather than a text-select for instance)

howard0su, 2019-02-11:
But look at the current UI code, it is pretty large on the ROM size
already. for example, PAGE_ScannerInit function, about 100 bytes for the
constants.

Maybe we don't need layout. we need a better way to handle the event. for
every page, we should feed all the event into one function, and let this
function to dealing with events, instead as today we define callbacks for
each control.

PhracturedBlue, 2019-02-11:
I'm certainly open to ideas.  I agree that it would be a lot better if we didn't have to custom-specify each control.