DeviationTX/deviation #683: deviationtx.com down

TheRealMoeder, 2019-02-11:
"This Account has been suspended."

PhracturedBlue, 2019-02-11:
working on it

TheRealMoeder, 2019-09-09:
Here we go again

PhracturedBlue, 2019-09-09:
I am aware.  Support is not being responsive.  I probably need to find a better hosting service, but don't have the bandwidth to do so right now

TheRealMoeder, 2019-09-10:
Issue solved, thanks @PhracturedBlue for once again quick response

TheRealMoeder, 2019-12-16:
Once again...

goebish, 2019-12-16:
Working fine here

PhracturedBlue, 2019-12-16:
works fine here too.  I changed the DNS entries, but that was 2 days ago and shouldn't affect anyone now.  We also have new firewall rules, but they shouldn't affect traffic directly to the website.

The return of nslookup and whatever error you browser returns may be helpful.

gdenton61, 2019-12-17:
http://www.deviationtx.com/ returns "404 Not Found" for me.
https works.

PhracturedBlue, 2019-12-17:
Thanks.  The redirect from www.deviationtx.com to deviationtx.com wasn't setup properly.  It should be working now.  Let me know if you still see issues.

closed 2019-12-18 by TheRealMoeder