DeviationTX/deviation #689: [UserVoice] WFLY protocol not working proper after poweroff

howard0su, 2019-02-13:
Repro steps:
1. turn on Rx and Tx
2. Binding with WFLY protocol
3. turn off rx and tx
4. turn on rx and tx
5. Rx is very lag.  

if Binding again, everything works again.

goebish, 2019-02-13:
I wish I could help but I don't have a WFLY Rx to check what's wrong, this protocol was ported from https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/WFLY_cyrf6936.ino

howard0su, 2019-02-13:
can we feedback to original author to help? I have Rx and if there is SPI dump, i can take a look as well.

goebish, 2019-02-13:
@pascallanger do you still have SPI dumps of this transmitter ?

pascallanger, 2019-02-13:
I've uploaded them here: https://we.tl/t-kULEKlAD38
It's strange since I'm not experiencing issues and I've got reports that it is working fine from different multi users as well...
Please keep me updated if you find something.

goebish, 2019-02-13:
A Deviation user also reported it was working fine when I ported it:
https://www.deviationtx.com/forum/protocol-development/7990-wfly-protocol

Perhaps there's some restrictions on usable frequencies or something like that ...

goebish, 2019-02-13:
@howard0su can you try with a different Fixed ID ?

pascallanger, 2019-02-13:
You could also try to use the TX parameters I've worked on using WFLY_FORCE_ID. This will help to see quickly where it fails due to a bad ID or something else.

howard0su, 2019-02-13:
thank you for your reply. I actually tried and not repro the issue. I will ask the user to try the suggestions.