DeviationTX/deviation #738: Bugs 3 callback

goebish, 2019-02-20:
https://github.com/DeviationTX/deviation/blob/0a5592c4dd2870f94bf56152f0ee91c0a63ef1be/src/protocol/bugs3_a7105.c#L644
Is this line useless or is it a bug?

hexfet, 2019-02-20:
I think neither.  If the count expires without completion, then return to the same state in 500us and again check if the operation is complete.  Contrast to hubsan_a7105.c line 501 where the code just continues as if completed.

It's arguable whether it would actually work if the operation is not complete on first try since the overall timing would be affected.

goebish, 2019-02-20:
The break exits the while loop, not the switch, packet_period is then set to DELAY_WAIT_RX

hexfet, 2019-02-21:
I see, good catch.  Same at line 579.  It could either be removed, or return instead of break.  Or remove the loop completely.  I can check if completion test ever fails.

closed 2019-02-22 by goebish