DeviationTX/deviation #785: Missing protocol for Jumper T8SG

maRCnh33, 2019-03-03:
I have some receivers with the ifhss protocol.... i would be nice if the protocol is available on my T8SG transmitter.....

i have also another issue. I have installed newest firmware yesterday (i was hoping that the ifhss protocol was included, but not...) But when i scroll true the protocols and i go back (to the first protocol at the list) the t8sg has a blank screen and the restarts itself!!! So i have tried different firmware versions but the same problem!! also when i am in the list with protocols and i push EXT it is restarting! So now i have installed the version that was on it before these updates. Version 1244 works the best! I am a little scared to update now because i do not want to lose my plane because of my tx is restarting itself....

goebish, 2019-03-03:
IFHSS isn't supported so far (never heard of it before). Can you open the transmitter and receivers and tell which ICs are inside ?
Regarding the problems you encounter with the nightly builds, that's normal, the firmware is under heavy rework and latest builds might not be stable.
For a stable version with all the latest protocols I think that's better to install vlad's 5.0 for now:
https://www.deviationtx.com/forum/builds/6407-deviation-v5-0-0-build-with-latest-protocols?start=460

maRCnh33, 2019-03-03:
I have checked the IC inside and this is it: 

CYRF6910
340LFXC
1307 A 04
CYP604991

the brand of the rx is : Ace RC

i hope you can help me with this! tnx!

goebish, 2019-03-03:
Ok, that's the programmable version (mcu+transceiver into a single chip) of the CYRF6936, pretty hard to crack :( I doubt we ever support this protocol.

goebish, 2019-03-03:
Can you check in the TX as well ? If we're lucky perhaps it's using a CYRF6936 with a separate MCU instead of a CYRF6910, in which case it could be more easily reverse engineered.

maRCnh33, 2019-03-03:
there is a CYRF6936 in my T8SG tx

goebish, 2019-03-03:
I mean in the original Ace RC Tx.

maRCnh33, 2019-03-04:
I only have some receivers... i dont have the original ace rc tx

goebish, 2019-03-04:
We can't do anything then, at least a transmitter is required to reverse engineer the protocol.

closed 2019-03-04 by goebish