DeviationTX/deviation #838: Devo 7e 3x2 switches not working

daniel-snowsurf, 2019-03-31:
With current version nightly & build from master the 3x2 switches are not working properly. Both switches SWA & SWB read wrong values:
0 position gets 0
1 position gets 0
2 position gets 1
Going back to Nov 2017 versions gets it working fine again. Details posted in:
[https://www.deviationtx.com/forum/6-general-discussions/8361-devo-7e-3x2-switches-not-working-anymore#73614](url)

PhracturedBlue, 2019-04-01:
Thanks for the report.  I've confirmed that while the 3x2 switches work fine with the devo7e-256, they do not work properly with the devo7e build.  I'm looking at it

vidzo, 2019-04-01:
I too am having the same issue using Moeders fork

TheRealMoeder, 2019-04-01:
Is this relaxed?
https://www.deviationtx.com/forum/stock-firmware-help/8366-t8sg-aux-5-not-working-after-fw-upgrade

PhracturedBlue, 2019-04-01:
I don't think so.  Likely recalibration is all that is needed for that issue.

closed 2019-04-03 by PhracturedBlue

vidzo, 2019-04-03:
Thanks. Switch is fixed, however Throttle trim TRIMLV- does not work. Furthermore RUDDER stick fails to work after adding 3x2 define in hardware.ini.

PhracturedBlue, 2019-04-03:
I can reproduce the trimLV- not working and I'll fix.  I cannot reproduce your rudder issue though.  My rudder works fine.  Try recalibration?

vidzo, 2019-04-03:
As movement isn't recognised, it won't calibrate that axis. If I remove the 3x2 define, channel/stick repsonds as standard. I'm using Moeders fork so not sure if there are significant differences

PhracturedBlue, 2019-04-03:
I cannot help you with Moeder's fork.

vidzo, 2019-04-03:
Ok, thanks anyway. I'll have to wait until voice alerts and editor/layout editor make it to the nightlies for Devo 7E

TheRealMoeder, 2019-04-03:
My feature branch doesn't touch this, but I can merge in current nightly, if it helps. Though I doubt this issue is related.

vidzo, 2019-04-03:
I'm not sure. I'll try again compiling with the changes, I don't see how it could break rudder channel but I'm no coder.

daniel-snowsurf, 2019-04-03:
Thanks for the fix. Just build the Devo 7e and 3x2 switches work fine. As already said LV trim is not working. I use it as a switch, it's always ON.

vidzo, 2019-04-08:
@Moeder, do you think you'll be able to look at the Rudder channel issue with the applied fixes to your master/voice_diet fork?

TheRealMoeder, 2019-04-08:
Do you see the rudder channel issue in current master?

vidzo, 2019-04-08:
It appears that I overlooked the master. It's all working fine, checked out to master, applied code and no issues with rudder. Sorry for not checking. So the issue is with voice_diet I think. No worries, I'll use master and probably end up making U7e. Thanks for your time and work Moeder and PB

TheRealMoeder, 2019-04-08:
I've updated voice_diet2 to current master...please report if the issue persists.

vidzo, 2019-04-16:
Another user has reported the same problem of being unable to use rudder once a calibration has taken place

TheRealMoeder, 2019-04-16:
I thought you didn't experience it anymore in master

vidzo, 2019-04-16:
I hadn't tested properly. I had flashed it, checked if 3x2 switch was working but didn't do a calibration of the sticks. Been involved with other projects so didn't check until the previous night, hence why I came back to GitHub to check

TheRealMoeder, 2019-04-22:
Rudder calibration issue on devo7e should be fixed with #862

PR #842 (2019-04-03): Fix 3x2 switch configu for devo7e
Fixes #838