DeviationTX/deviation #855: switchs A/B/C/D/E/F/G/H doesn't work on Emulator T8SG V2

flagadajones, 2019-04-12:
none of the switchs on the t8sg v2 work on the emulator.
but I think the problem is not limited to this target.

albfan, 2019-06-26:
That's because they are unmapped. You need to go to mixers and activate those like:
```
- channel 5 = simple aux40
- channel 6 = simple aux5
- channel 7 = simple swA0
- channel 8 = simple swB0
...
```
using jstest-gtk (linux)
What I found strange is that it recognizes 8 axis (I can only imagine 6 including aux 4/5) and only 4 buttons

[image]

```
$ sdl-jstest --list
Found 1 joystick(s)

Joystick Name:     'DeviationTx Deviation GamePad'
Joystick Number:    0
Number of Axes:     8
Number of Buttons:  4
Number of Hats:     0
Number of Balls:    0
```

The correct setup would be: 6 axes and 10 buttons. With 10 buttons, tristate buttons like swA would do 
`swA0` as one button and `swA2` as other (so `swA1` is like off all those buttons)

- 4 tristate buttons x 2 = 8
- normal buttons x 2 = 2

Not sure how to setup that, just have a flysky fs-i6x and it maps all those tristate buttons. If anyone has a clue about what to change to map /dev/input/js0 correctly I can work on it.

albfan, 2019-06-30:
Seems right now it will only support up to 8 channels for buttons

https://github.com/DeviationTX/deviation/blob/master/src/protocol/usbhid.c#L27

so it will need some work to cover 10 buttons.

Same for axis, 8 is just a big enough number

albfan, 2019-06-30:
@howard0su would a patch to customize axes and buttons be accepted? All supported models can use a default one (8 axes, 4 buttons) until someone customizes it.

If you have more details on what is needed to support more than 8 buttons, that would be appreciated

albfan, 2019-07-01:
I started working on these. About setup, I'm using docker. The best way I found is to go into shell and execute:

```
cd /git/deviation/src
make TYPE=prd TARGET=t8sg_v2_plus
```

Using GUI from docker it builds without problem, but using shell it ask me to install cpplint. I installed with:

```
sudo apt-get install python3-pip
sudo pip3 install cpplint
```

But not sure what is different. I'm checking deviation-docker to understand it.

about flash the transmitter, we enter in DFU mode pressing go-back and power. Seems:

```
$ dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [0483:df11] ver=0200, devnum=6, cfg=1, intf=0, path="2-1.2", alt=2, name="@SPI Flash: Library/0x00020000/030*064Kg", serial="DEVO-7E ??????C"
Found DFU: [0483:df11] ver=0200, devnum=6, cfg=1, intf=0, path="2-1.2", alt=1, name="@SPI Flash: Config/0x00002000/030*04Kg", serial="DEVO-7E ??????C"
Found DFU: [0483:df11] ver=0200, devnum=6, cfg=1, intf=0, path="2-1.2", alt=0, name="@Internal Flash  /0x08003000/00*001Ka,244*001Kg", serial="DEVO-7E ??????C"

dfu-utils -a 0 -D file.dfu
```

should be enough, I cannot make a backup since -U option limit to 16384

```
Device returned transfer size 1024
Limiting default upload to 16384 bytes
Upload	[=========================] 100%        16384 bytes
Upload done.
```

I think I could flash from dfu of last zip in case of problems

About upload files, seems enter in USB is enough altough power with enter pressed is another option

My idea is to enable more than 8 buttons, but not sure how to make that portable (if that implies a problem with final dfu size)

somewhatlurker, 2020-09-17:
Oh shoot...  this probably was never even related to USB; just that there seems to be no way to control those switches in the deviation emulator.

I think that's caused by code in `deviation/src/target/drivers/input/addon_switch/` being used for hardware only and `deviation/src/target/drivers/mcu/emu/channels.c` making no effort to support extra switch config.

Probably not really worth fixing though when the devo10 and devo12e emulators should support everything.

TheRealMoeder, 2020-09-17:
Always nice to have everything in emulators working, but as you said probably not really worth the effort. There are other things which are more important.

albfan, 2020-09-20:
@somewhatlurker What is exactly missing. I think it is covered but only partially. isn't it? 

@TheRealMoeder any contribution on this will be accepted? Any hints on how to cover this feature? represent correctly tristate buttons would be great, I can dig into joystick spec to see how to do that propertly.

somewhatlurker, 2020-09-20:
In deviation emulators for t8sg v2 (at least -- possibly others too) there seems to be no keyboard key to toggle any switches except generic HOLD and FMODE ones. So A,B,C,D,G,H are completely unavailable.
The emulator's switch input code doesn't seem to understand switch configuration like the code for real hardware does.

albfan, 2020-09-20:
@somewhatlurker reread https://github.com/DeviationTX/deviation/issues/855#issuecomment-505838274

They come unmapped, but you can map it.

The problem is you cannot map all of them because the model only allows 8 maps (one byte) and you need two for each tristate.

6x2 = 12

But I think I have all mapped, so maybe I use virtual, have to check my config.

I'm using some of them to activate OSD overlay, recover from crash, acro mode, so they work for sure.

somewhatlurker, 2020-09-20:
Emulator in this case probably isn't using a transmitter as a controller though. There are "emulator" versions of the deviation firmware that run on a computer for testing it or configuring settings without using a transmitter.

Since the switches literally don't work there as far as I can tell (and the op mentions E/F which don't really exist on an unmodified t8sg v2), that's probably what the original issue meant.

somewhatlurker, 2020-09-20:
I agree that it's worth adding more buttons (though iirc deviation only runs 16 mixers?? so 8 axis and 8 buttons is probably a sane limit), but it seems like this is two separate issues.

albfan, 2020-09-20:
Ah ok, https://www.deviationtx.com/manuals/html-devo10/ch_emulator.html

Didn't know emulator was a thing