DeviationTX/deviation #858: Screen dim on late-model T8SG

jkflying, 2019-04-14:
I flashed latest nightly, and it seems the screen-dim issue has returned as documented [here](https://www.deviationtx.com/forum/stock-firmware-help/7559-jumper-t8sg-blank-screen?start=40).

I've flashed back to a slightly older version, https://github.com/DeviationTX/deviation/commit/0751c6fac882e42a3069acfcd9cfd527e85fb222 , and the issue is fixed. It is a real pain to bisect this, since I don't have all of the builds and older builds are difficult with the existing docker images since the file structure changed, but I suspect the issue came from this merge https://github.com/DeviationTX/deviation/pull/834 which did a lot of work on unifying the LCD drivers, which of course makes it easy to lose some sub-model specific tweaks.

I'm happy to test new builds.

vladislavy, 2019-04-17:
I can't find LCD_Cmd(0x26); //Select Internal Resistor Rate (Rb/Ra) for T8SG V1 and V2 _lcd_init(), devo7e LCD use default value for that.

vladislavy, 2019-04-19:
You can try the next nightly build, it has to be fixed.

jkflying, 2019-04-22:
Confirmed this is fixed, thank you!

closed 2019-04-22 by jkflying