DeviationTX/deviation #866: Flashing Radiolink AT9

SilverFire, 2019-05-02:
Hi everyone.

Yesterday I've tried to build DeviationTX for my Radiolink AT9 and I have experienced some troubles, so would like to share my experience of the upgrade, DFU bricking and recovery procedure.

# Buiding DeviationTX for Radiolink AT9

I've used Docker image to build firmware from sources. The [building toolkit](https://www.deviationtx.com/wiki/development/docker) does not suggest an option to build for AT9, but I've created a PR https://github.com/DeviationTX/deviation-docker/pull/7. Most probably it will not be accepted in the nearest future as AT9 is not stable enough even for nightly builds. But you can always escape building UI and run

```bash
sudo /root/build.py --manual-prereq
cd /git/deviation/src
PATH="$PATH:/opt/gcc-arm-none-eabi-8-2018-q4-major/bin"
sudo make zip_at9
mv deviation-at9* build/
```

Take built ZIP archive, unpack it and get ready to flash

# Faulty flash

First, I've tried to flash AT9 as it was suggested [here](https://www.deviationtx.com/forum/7-development/5183-radiolink-at9-as-a-platform-for-deviation?start=140#55023), but after success flashing of `deviation-at9-v5.0.0-********.dfu`, when I've tried to flash fs, the Radiolink official flashing tool exited after pressing "Download" without any success.

The transmitter turned on with black screen and power LED and was displayed as Radiolink AT9 is Zadig when connected with USB.

The  `update.Log` of Radiolink flashed containd the error:

```
[main_dfu @1162] dif->alt_name:@Internal Flash  /0x08000000/12*001Ka,500*001Kg
[main_dfu @1172] Claiming USB DFU Interface...
[main_dfu @1179] Setting Alternate Setting #0 ...
[main_dfu @1187] Determining device status: [main_dfu @1194] state = dfuERROR, status = 10
[main_dfu @1206] dfuERROR, clearing status
error clear_status
```

<details>
  <summary>Click to see full log</summary>

```
[print_version @655] dfu-util 0.6

[print_version @658] Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY

[main_dfu @899] Filter on vendor = 0x0483 product = 0xdf12
[main_dfu @923] Find Device ...
[iterate_dfu_devices @390] bus-0/\\.\libusb0-0001--0x0483-0xdf12     0483/DF12
[main_dfu @940] get_first_dfu_device
[iterate_dfu_devices @390] bus-0/\\.\libusb0-0001--0x0483-0xdf12     0483/DF12
[main_dfu @947] Opening DFU capable USB device... 
[main_dfu @961] get_first_dfu_if
[find_dfu_if @168] idVendor:483
[find_dfu_if @169] idProduct:df12
[find_dfu_if @178] bConfigurationValue:1
[find_dfu_if @190] bInterfaceClass:fe
[find_dfu_if @191] bInterfaceSubClass:1
[find_dfu_if @192] bInterfaceNumber:0
[find_dfu_if @193] bAlternateSetting:0
[find_dfu_if @194] bInterfaceProtocol:2
[main_dfu @966] ID 0483:df12
[main_dfu @974] get_cached_extra_descriptor
[get_cached_extra_descriptor @600] bus-0/\\.\libusb0-0001--0x0483-0xdf12     0483/DF12
[get_cached_extra_descriptor @607] num_altsetting = 2
[get_cached_extra_descriptor @608] extra = (null)
[get_cached_extra_descriptor @609] extra_len = 9
[get_cached_extra_descriptor @612] extra:(null) [bLength:9]
[main_dfu @992] Run-time device DFU version 0000
[main_dfu @1135] count_matching_dfu_if
[find_dfu_if @168] idVendor:483
[find_dfu_if @169] idProduct:df12
[find_dfu_if @178] bConfigurationValue:1
[find_dfu_if @190] bInterfaceClass:fe
[find_dfu_if @191] bInterfaceSubClass:1
[find_dfu_if @192] bInterfaceNumber:0
[find_dfu_if @193] bAlternateSetting:0
[find_dfu_if @194] bInterfaceProtocol:2
[find_dfu_if @190] bInterfaceClass:fe
[find_dfu_if @191] bInterfaceSubClass:1
[find_dfu_if @192] bInterfaceNumber:0
[find_dfu_if @193] bAlternateSetting:1
[find_dfu_if @194] bInterfaceProtocol:2
[main_dfu @1150] get_matching_dfu_if
[find_dfu_if @168] idVendor:483
[find_dfu_if @169] idProduct:df12
[find_dfu_if @178] bConfigurationValue:1
[find_dfu_if @190] bInterfaceClass:fe
[find_dfu_if @191] bInterfaceSubClass:1
[find_dfu_if @192] bInterfaceNumber:0
[find_dfu_if @193] bAlternateSetting:0
[find_dfu_if @194] bInterfaceProtocol:2
[main_dfu @1162] dif->alt_name:@Internal Flash  /0x08000000/12*001Ka,500*001Kg
[main_dfu @1172] Claiming USB DFU Interface...
[main_dfu @1179] Setting Alternate Setting #0 ...
[main_dfu @1187] Determining device status: [main_dfu @1194] state = dfuERROR, status = 10
[main_dfu @1206] dfuERROR, clearing status
error clear_status
```

</details>

I've tried every single driver mode in Zadig 2.4, tried to change USB cable - nothing helped me.  Then I've tried to flash manually with the `dfu-util`, that seemed to work under the hood of Radiolink flasher, but have seen the same error.

Soon, I've found [the article](http://kp1533tm2.blogspot.com/2016/05/radiolink-at9-firmware-update-on-linux.html) that pointed out on error in `dfu-util` that leads to exactly the same error I have, but it is fixed in `master` branch and is not released yet (for more than 2 years, sadly). I've compiled `dfu-util` from sources but saw the same error.

```
$ ./dfu-util -vv -a 0 -i 0 -s 0x08000000 -D ../deviation-at9-v5.0.0-d9a11bb.dfu
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2019 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DFU suffix version 11a
Match vendor ID from file: 0483
Match product ID from file: df12
Opening DFU capable USB device...
ID 0483:df12
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
error clear_status
```

(at this point I've already tried to restore original firmware.

# Recovery after the faulty flash

It was obvious for me that DFU loader is somewhat broken, so I've soldered pins to serial pads on the AT9 mainboard and connected USB > UART TTL (CH340) adapter as follows

| CH340 | AT9 |
|----|----|
| TX | RX |
| RX | TX |
| GND | GND |
| DTR | BOOT0 |
| -- | 3V3 |

[image]

1. Download [STM Flasher](https://www.st.com/en/development-tools/flasher-stm32.html) and [DfuSe](https://www.st.com/en/development-tools/stsw-stm32080.html) and instal both
2. Download original Radiolink firmware from [Radiolink website](http://www.radiolink.com.cn/doce/at9upgrade.html)
3. Run DFU File Manger, and extract `s19` from the original `*.dfu` file
4. Connect and turn on CH340, plug in AT9 power, start Flash Loader Demonstartor, connect to STM32 and UPLOAD firmware to your PC from a device as a backup (actually, it must be the first step before I tried to flash something :) good lesson for me though)
5. Reset power, start Flash loader again, choose `Download to device`, select original `*.s19` file you've extracted from the original FW, set `Erase necessary pages` and `Verify after download`. Flash.
6. Disconnect power, CH340 and reboot the transmitter.
7. Congrats, the firmware is restored!

# Flashing Deviation with serial port

Is mostly the same as flashing original firmware.

1. Build
2. Extract `.p19` from both `deviation-at9-v5.0.0-*******.dfu` and  `at9-fs.dfu` files
3. Flash both with the same options as original FW.
4. Disconnect everything and there you go!

[image]

# Flashing Deviation with DFU

Did not tried any more

# My experience

1. Controls: sticks, buttons, knobs and so on are mapped correctly and work as expected.
2. Screen: looks as expected
3. Crashes: trying to change a model name takes transmitter in a bootloop, only re-flashing helps
4. Radio: trying to scroll over the supported protocols brings TX in a bootloop after showing PXX option

# Summary

I have ordered new transmitter with OpenTX support so tried to play a game with my old AT9. It was fun, but I don't have enough experience to debug SMT32 and contribute to your project, however, I can help you with proper testing of AT9 build, if you have eager to make it work. Otherwise, this issue can be safely closed with `won't fix` tag :) You've done a great job anyway!

SilverFire, 2019-05-02:
Oh, and for those, who have succeeded to brick AT9 completely, so you can not restore firmware as described above. Here is a full dump of all pages of STM32 with working original firmware v.1.7.8 and DFU. Flash it with the `Global erase` option checked. But don't forget to make a backup of your bricked firmware, maybe you will need it for some reason.

[Radiolink_AT9_DFU_v1.7.8.zip](https://github.com/DeviationTX/deviation/files/3139975/Radiolink_AT9_DFU_v1.7.8.zip)

sawyouintheheaven, 2019-05-17:
Thank you@SilverFire!I burned my AT9 due to wrong battery polarity.I don't know how many ICs burned out.Maybe the MCU ,the EEPROM or the 2.4G module. But I'm sure the mcu has been dead.And I replaced it with a new one.And I follow your procedure to  flash the original FW. But when I switch on my AT9,it just keep beeping.There seems to be a self-test mechanism,maybe the program check the EEPROM or the 2.4G module. I'm sure the FW was flashed well because the AT9 is recognized when I connect it to my PC.
Although I haven’t repaired it yet,thank you very much！
I'm trying to ask to the customer service.But they have not reply yet .Could you please help me to read out data from the EEPROM?

SilverFire, 2019-05-18:
The EEPROM marked as `ST 449 4128a2gl` seems to be [this one](https://www.st.com/resource/en/datasheet/m24128-bf.pdf). It's a 128k I2C EEPROM. I've made a dump for you. It contains my model settings, but nevertheless, it works.

[Radiolink_AT9_EEPROM_128k.zip](https://github.com/DeviationTX/deviation/files/3193807/Radiolink_AT9_EEPROM_128k.zip)

sawyouintheheaven, 2019-05-18:
Oh,thank u so much for your continued help!I will buy this EEPROM and flash it.