DeviationTX/deviation #874: no telemetry T8SG plus Frsky FPort

mooiweertje, 2019-05-28:
I'm using an Frsky XSR receiver on FPort but only get rssi telemetry. The FC telemetry is lost somewhere.

hexfet, 2019-05-28:
This should be fixed by PR #875.  Please try the test build [here (219d660)](https://www.deviationtx.com/downloads-new/category/1725-hexfet-sport-fix) .

mooiweertje, 2019-05-28:
I installed it and I've got 1 battery cell now. So communication is working.
But the other fields, like GPS data are still empty.

hexfet, 2019-05-28:
I use a sensor simulator to feed data to my XSR and all the sensors are being displayed correctly on my T8+.  Do you have a way to check the data on the f.port connection to be sure the expected sensors are being sent?

mooiweertje, 2019-05-29:
I tried with an FrSky transmitter and the same issue rises. So the problem
seems to be on the XSR/FC side now.

thanks for your help.

hexfet, 2019-05-29:
Thanks for reporting back.  Good luck.

closed 2019-05-29 by hexfet

mooiweertje, 2019-06-03:
I guess there is no relation with installing this test version, but...
My T8SG died the next day. Pressing the powerbutton doesn't swich on the RC
anymore. I have a simple video showing the issue. I do hear a tiny squeeck
from the speaker I build in but it's too soft to record. I think the
mainboard is bricked somehow.

Do you have any idea what I can do?

I have contacted the seller to send me a new mainboard but the are not very
cooperative.