DeviationTX/deviation #877: I8X-IR8M missing module A7105

dheku, 2019-05-30:
Hi.I use i8x-ir8m.

2 months ago, I updated firmware to deviation-ir8m-v5.0.0-2455ad1.

Problem has occurred so it indicated alert as missing modules A7105.

I confirmed hadrware.ini but it was not incorrect that enable A7105.

I want to fix this problem.

The problem was solved after changing firmware to deviation-ir8m-v5.0.0-b08efa8.

And it is not filesystem problem

PS:Recently deviation-ir8m-v5.0.0-9fd50b5 has same problem.

metalstrolch, 2020-08-30:
Since I recently got a IR8-M I was looking after this since I found no working firmware supporting WFLY protocol.

Bisecting the problem, I found this was introduced by #749 by moving the PWR_SWITCH_PIN and PWR_ENABLE_PIN initialisation after DisableJTAG(); in src/target/drivers/mcu/stm32/power.c .
However I do not understand now why. If I move DisableJTAG() down again, even current HEAD runs without this error.

Unfortunately WFLY protocol seems to be broken on HEAD. It works indeed on

metalstrolch, 2020-08-30:
See https://github.com/metalstrolch/deviation/tree/ir8m-missing-A7105-fix for mentioned patch. Don't know exactly why this is the fix, and don't know exactly if this could have effects on other TX.