DeviationTX/deviation #879: ENT button do not work properly on Devo10

smaller09, 2019-06-06:
In order to use the voice alarm on my Devo10 , I upgrade the firmware from vlad_vy's 5.0.0 build to the newest nightly.(includeing the filesystem from nightly to prevent not compatible)

  After that, the voice alarm works like a charm, but soon I found the ENT button works incorrect.
In several submenu: "model load", "model name edit" ,  "protocol setting", the ENT button functions like the L button, it can not enter, but change selections or move the cursor left.
that means I can not change modle or protocol setting. 

  I went back several commits and recompile the firmware, and found the bug was first introduced by commit "Cleanup _tr usage (DeviationTX#794)" and then by commits “[WIP] Consolidate swtch code 4/4 (DeviationTX#826)， Remove channels.c for all transmitters (DeviationTX#824)，[WIP]Consolidate switch code 2/4 (DeviationTX#820)  [WIP]Consolidate switch code 1/4 (DeviationTX#819) ”

  I forked my own repository, bypass those commits and several commits relates to button input.
and finally got a working firmware for my Devo10, but I still need to fly to confirm.

hexfet, 2019-06-07:
Just tested the latest nightly build (28ce50f) on my Devo10.  Did not have any issues with the ENT button.  It worked as expected.  Tested with new filesystem, and with voice enabled and disabled (though I don't have voice hardware in the 10).

If you installed the audio module internally maybe something's going on with the hardware?
Maybe post a video showing the symptom.

smaller09, 2019-06-10:
I had tested my devo10 yestoday， it works find with my own builded firmware. 
I just flashed the latest nightly build (28ce50f), the problem was still there. and I was not the first one to report this problem.
[https://www.deviationtx.com/forum/6-general-discussions/8436-devo10-ent-button-problem](url)
and I took a video of that. 
<iframe height=498 width=510 src='http://player.youku.com/embed/XNDIyMjgyODY2MA==' frameborder=0 'allowfullscreen'></iframe>

hexfet, 2019-06-10:
Thanks for testing.  I tried another devo10 and the devo10 emulator.  Still unable to replicate the problem.

One strange thing in the video - Afrikaans appears twice in the language list.  That should not happen in a standard install.  If you haven't done that on purpose, I recommend formatting the USB drive and re-installing the filesystem.  Should be fine to save/restore tx.ini, hardware.ini, models and icons, but please test for the ENT error before making any changes to the vanilla install.

I looked at PR #794 and it makes changes for language translation.  Did not spot a change in there that would affect the ENT key behavior.  Without being able to replicate the problem I can't do much but look at the code.  If you're willing to do some debugging please use git bisect to identify the first commit where the problem appears.

smaller09, 2019-06-11:
I was also notified the two Afrikaans language item( I thought it was from my old backup). and had deleted one, but it did not help things better.
And I had already tried using DeviationUploader-0.9.0.jar to flash the whole nightly zip file to prevent not compatible, the problem was still there. I can do again and send a video later.
As for git bisect. I had already done this, I had try went forward commits one by one to find the problem. and pretty sure that the problem was first introduced by commit "Cleanup _tr usage (#794)", and I can use git bisect to make sure again.
One more thing, the problem was not actually happend anytime. after a clean flash. the ENT buttom have a chance to work. but after did some setting or reboot the transmitter, it comes back. I through  the problem maybe relates to the file saving, but I don't have the ability to check the code.

hexfet, 2019-06-12:
It is a strange issue.  PR #834 only changes the display driver code.  Don't see anything there that would affect button inputs.  With the other symptoms you describe it would not surprise me if running bisect again might land on a different commit.

What kind of battery is in the devo10?  Maybe and intermittent voltage issue.
If it works after a clean flash, if you don't change anything and turn the transmitter off and back on does it still work correctly?

smaller09, 2019-06-13:
OK, I have another modfree Devo 10 to test, with the lastest nightly, I works good.
So the problem my be with my mods.
But I could not understand why my own builded firmware works correct.

KampfGurke20, 2019-06-15:
I have the same Problem here, Devo 10 with 4in1 Module :(

Could you upload your modded Firmware?

smaller09, 2019-06-16:
> I have the same Problem here, Devo 10 with 4in1 Module :(
> 
> Could you upload your modded Firmware?
Please try it, only reverted PR#834. any other things updated to the main repository
[https://github.com/smaller09/deviation/releases/download/v5.0.0_ent_fix/deviation-devo10-v5.0.0-658fd40.zip](url)

KampfGurke20, 2019-06-16:
Thanks i will try that!

I found Out that If i Push the Ent and the Down Button at the Same Time i could enter the Modell Menu and select another One. (Latest nightly)

BrainHunter, 2019-07-14:
The Devo i tested with was unmodified! 
--> https://www.deviationtx.com/forum/6-general-discussions/8436-devo10-ent-button-problem

If you doubleclick the ENT button, it sometimes work.
In the git log i saw some changes to the switch mux a while ago. -> maybe a timing issue?

vladislavy, 2019-12-20:
I can't reproduce it with any my transmitter. I haven't Devo 10. Can you say, if you exit problem page, will ENT button work normally on other pages?

vladislavy, 2020-03-24:
devo10 buttons fixed.

closed 2020-03-24 by vladislavy