DeviationTX/deviation #883: t8sg v2 plus

ariefadha, 2019-06-20:
i hop this is the right place.

deviation-t8sg_v2_plus-v5.0.0-28ce50f 

here is the bug that i found
1. to turn off the TX its no longer need to hold the button.
2. timer stop-prop and cntdn-prop, control switch not effected the timer

Tygrys-1, 2019-08-07:
1) Confirm, it should be longer.
3) It is the feature, not a bug. Left button is for reset and right for start/stop if no other switches are assigned for these functions in the timer configuration.

TheRealMoeder, 2019-08-28:
I don't quite understand the second issue

ariefadha, 2019-08-29:
for the second issue : if i fly "plane" or "drone" i want the cntd-prop to count when only the props are spinning, or have a switch that can pause the cntd-prop-timer. how i do that?

Tygrys-1, 2019-08-29:
If you use a 3-pos switch, in the middle position the proportional counter will stop.
For throttle-dependant see #887 - this will measure the time proportional to the THR value. 
You may also use the normal stopwatch timer with the switch set to THR - it will start count at normal speed for the throttle value >-90

vladislavy, 2019-12-18:
Fixed

closed 2019-12-18 by vladislavy